PlotJuggler/PlotJuggler #861: Plotjuggler compilation issue with FAST-CDR 1.1.0

MMarcus95, 2023-08-25:
Thanks for contributing to PlotJuggler. You are great!

# Problem description
Hello, I'm trying to compile Plotjuggler from source but it is not able to find fastcdr package. The installed version of FAST-CDR is 1.1.0. When compiling Plotjuggler (using either the last commit on main or the latest version 3.7.1) I'm getting the following error

[image]

I'm actually having this issue since FAST-CDR 1.0.27. Since FAST-CDR 1.1.0 is the suggested version for FAST-DDS 2.11.0, it could be really nice to have Plotjuggler compiling with this version.

# Steps to reproduce (important)

- Platform: Ubuntu 22.04
- Install FAST-CDR 1.1.0
  - git clone https://github.com/eProsima/Fast-CDR.git -b 1.1.0
  - mkdir Fast-CDR/build && cd Fast-CDR/build
  - cmake .. && cmake --build . --target install
- Install Plotjuggler as described [here](https://github.com/facontidavide/PlotJuggler/blob/main/COMPILE.md#compile-in-linux)

paulBismuthListen, 2024-01-19:
Hello, 
Hhad the same on oracle 8 distribution.
Look like it is gone with cmake udpate, version I used is cmake-3.26.5

cheers.

closed 2024-01-26 by facontidavide

facontidavide, 2024-01-26:
minimum cmake version I MUST support is 3.16.3, to be compatible with Ubuntu 20.04