PlotJuggler/PlotJuggler #862: Building DataLoadParquet on Windows

ngbrown, 2023-08-28:
Thanks for contributing to PlotJuggler. You are great!

# Problem description

I am trying to build the DataLoadParquet plugin on Windows, but there are no clear instructions on what the correct configuration of dependencies is.

# Steps to reproduce (important)

I installed `arrow` with `.\vcpkg.exe install arrow:x64-windows` but when building PlotJuggler, it didn't build the DataLoadParquet plugin. The following was in the cmake generate log:

```
-- Finding Parquet with vcpkg
-- Arrow version:
-- Found the Arrow shared library: D:/dev/OSS/PlotJuggler/build/vcpkg_installed/x64-windows/bin/arrow.dll
-- Found the Arrow import library: D:/dev/OSS/PlotJuggler/build/vcpkg_installed/x64-windows/lib/arrow.lib
-- Found the Arrow static library:
[Parquet] not found. Skipping plugin DataLoadParquet.
```

Are there steps I'm missing?

- Windows 10 22H2

ngbrown, 2023-08-29:
The output of `.\vcpkg.exe install arrow:x64-windows` includes the following lines:

```
find_package(Arrow CONFIG REQUIRED)
target_link_libraries(main PRIVATE "$<IF:$<BOOL:${ARROW_BUILD_STATIC}>,Arrow::arrow_static,Arrow::arrow_shared>")

find_package(Parquet CONFIG REQUIRED)
target_link_libraries(main PRIVATE "$<IF:$<BOOL:${ARROW_BUILD_STATIC}>,Parquet::parquet_static,Parquet::parquet_shared>")
```

I noticed that the `CMakeLists.txt` uses lower case `arrow`, so I changed the following line to `find_package(Arrow CONFIG QUIET)` and it worked:

https://github.com/facontidavide/PlotJuggler/blob/078d2e8a8e0236749c89bfe75422f5417630b7cb/plotjuggler_plugins/DataLoadParquet/CMakeLists.txt#L9

The next problem was:

```
  Automatic MOC and UIC for target DataLoadParquet
..\..\vcpkg_installed\x64-windows\bin\parquet.dll : fatal error LNK1107: invalid or corrupt file: cannot read at 0x2F8
[D:\dev\OSS\PlotJuggler\build\plotjuggler_plugins\DataLoadParquet\DataLoadParquet.vcxproj]
```

To fix this, I undid #807 and used `PARQUET_LIBRARIES` from the `BUILDING_WITH_VCPKG` instead of:

https://github.com/facontidavide/PlotJuggler/blob/078d2e8a8e0236749c89bfe75422f5417630b7cb/plotjuggler_plugins/DataLoadParquet/CMakeLists.txt#L38

And changed the `BUILDING_WITH_VCPKG` from:

https://github.com/facontidavide/PlotJuggler/blob/078d2e8a8e0236749c89bfe75422f5417630b7cb/plotjuggler_plugins/DataLoadParquet/CMakeLists.txt#L1-L11

to:

```cmake
if(BUILDING_WITH_VCPKG)
    message(STATUS "Finding Parquet with vcpkg")
    set(PARQUET_LIBRARIES
        Arrow::arrow_shared 
        Parquet::parquet_shared)

    find_package(Arrow CONFIG QUIET)
    find_package(Parquet CONFIG QUIET)

else()
...
```

Evidently this conflicts with linking to Parquet without package managers, thus #807. I don't know what the fix for that conflict is.

Vadoola, 2023-10-02:
I'm curious as well I wanted to build this and my initial couple of attempts failed, but I never had much time to dig into this. It was a few months ago I tried, so I don't remember the exact error. I may need to try again and see if it's the same as your getting.

auxym, 2024-10-28:
Also, it seems that the Windows installer for PlotJuggler does not include the parquet plugin?

Vadoola, 2024-10-28:
> Also, it seems that the Windows installer for PlotJuggler does not include the parquet plugin?

It doesn't not, that was the main reason I was trying to build it, was just to use it. I never managed to dig into it anymore. I realized while my main machine I'm using PlotJuggler on is Windows, I do have a Linux VM, so I just installed PJ on that with the parquet plugin looked at what I needed and that was that.

It would be nice if this plugin could be bundles with the Windows installer.

facontidavide, 2026-09-20:
PlotJuggler 4 replaced the vcpkg-based Parquet build with a Conan-provided Arrow/Parquet setup (find_package(Arrow/Parquet), linked as static targets), so the case-sensitivity and parquet.dll LNK1107 problems here no longer apply. Windows installer bundling is tracked separately as part of the deferred Windows release target; closing this build-config bug as obsolete.

---
*PlotJuggler 4 is now the official release and the default branch of this repo; PlotJuggler 3 is end-of-life and no longer receives features or fixes. If this still affects you in PJ4, please reopen or file a fresh issue — it will be looked at.*

closed 2026-09-20 by facontidavide