PlotJuggler/PlotJuggler #864: Loading layout file with spit panel always causes segmentation fault

scrice, 2023-08-29:
First of all, love this tool. I'm excited to use it.

I have noticed that if I try to load even the most basic of layout files with a horizontal or vertical split of any kind, plotjuggler crashes. This is when running the ros2 package. This is a pretty critical issue as it means I have to manually recreate layouts every time I run plotjuggler if I want anything more than a single plot panel (tabs do still seem to work though).

Running the command:
ros2 run plotjuggler plotjuggler --layout test_layout.xml

The same behavior happens when loading the test layout file through the gui.

[image]

I am running ros2 humble on ubuntu 22.04 on wsl.

duane-impulse, 2023-10-11:
I experience the same behavior. Similar setup, but not using WSL.

fbottarel, 2023-10-16:
Same issue, with PlotJuggler running in a minimal ubuntu 22.04 docker.

mgiglia92, 2023-11-12:
Having the same issue here, seems to only crash on splits for me as well.

```
Stack trace (most recent call last):
#31   Source "./csu/../sysdeps/nptl/libc_start_call_main.h", line 58, in __libc_start_call_main [0x7f7a18c29d8f]
#30   Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x55909e8d5686, in 
#29   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f7a196c0cf3, in QCoreApplication::exec()
#28   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f7a196b875a, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
#27   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f7a197130b7, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#26   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f7a18e8a3e2, in g_main_context_iteration
#25   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f7a18ee2257, in g_io_channel_new_file
#24   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f7a18e8cd3a, in g_main_context_dispatch
#23   Object "/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5", at 0x7f7a154f8d6d, in QXcbNativeInterface::dumpNativeWindows(unsigned long long) const
#22   Object "/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f7a19b16a2b, in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#21   Object "/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f7a19b41306, in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*)
#20   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f7a196b9e39, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#19   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f7a1a36c712, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#18   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f7a1a3cbfd4, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#17   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f7a1a3c8d3f, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#16   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f7a1a372e46, in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool)
#15   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f7a196b9e39, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#14   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f7a1a374363, in QApplication::notify(QObject*, QEvent*)
#13   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f7a1a36c712, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#12   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f7a1a3af4ed, in QWidget::event(QEvent*)
#11   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f7a1a4641e6, in QAbstractButton::mouseReleaseEvent(QMouseEvent*)
#10   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f7a1a463fc3, in QAbstractButton::nextCheckState()
#9    Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f7a1a46239d, in QAbstractButton::hitButton(QPoint const&) const
#8    Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f7a1a462115, in QAbstractButton::clicked(bool)
#7    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f7a196f14e4, in QObject::setProperty(char const*, QVariant const&)
#6    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x55909e8e1169, in 
#5    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x55909e8de951, in 
#4    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x55909e9161a8, in 
#3    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x55909ea72bcc, in 
#2    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x55909ea6ae5d, in 
#1    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x55909ea6aa79, in 
#0    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f7a1955e11b, in QStringRef::toDouble(bool*) const
Segmentation fault (Address not mapped to object [0x55900000025b])
[ros2run]: Segmentation fault
```

Thomas-Merlet, 2023-12-05:
Same issue for me on Ubuntu 22.04 LTS, ROS2 Humble, and PlotJuggler 3.7.1

closed 2023-12-20 by facontidavide

facontidavide, 2023-12-21:
fixed

commit 4e9cff9a (2023-12-20): critical bug fix: #864 #856