PlotJuggler/PlotJuggler #866: Plotting values without connecting them with a line but as a step

marios-stam, 2023-09-11:
Hello all and thank you for this amazing software.

i was wondering if there is feature or somehow can be implemented to plot data without connecting 2 successive points with just a line   but have a step effect on the plotting. Something similar to the below:
[image]
 
Right now the plot is like the left and want to make it like the right.

Thank you very much!

marios-stam, 2023-09-13:
I am sorry for not mentioning it, although I think you will notice either way, that these specific plots were made through pyplot just to showcase what I am trying to say.

zdavkeos, 2023-11-02:
This can be closed since #869

closed 2024-04-02 by marios-stam