PlotJuggler/PlotJuggler #878: 'No parser available for encoding [cdr] nor [ros2msg]' after mcap conversion from ros2 db3 bag

lucasw, 2023-10-24:
Thanks for contributing to PlotJuggler. You are great!

# Problem description

I converted a ros2 db3 to mcap using `mcap convert` which results in this error when trying to load in Plotjuggler:

```
No parser available for encoding [cdr] nor [ros2msg]
```

Then similar for a ros1 bag:
```
No parser available for encoding [ros1] nor [ros1msg]
```

The mcap files report as 

```
mcap info *mcap
library:   mcap go v1.0.4                                              
profile:   ros2                                                        
```

That's using a version of the mcap cli tool built from latest source (should I go to an older version, a release like https://github.com/foxglove/mcap/commit/91b3a9f41ee967d079ce964119b6f004e97d8c0e ?).

# Steps to reproduce (important)

* Ubuntu 23.04
* Plotjuggler built from latest source a123c36350b4ef116c7d76a79c16832eec6f9643
* ros2 iron built from latest source

```
git clone git@github.com:foxglove/mcap.git
cd mcap/go
make -C cli/mcap build
# mcap binary is now mcap/go/cli/mcap/bin/mcap
```

Using data from https://github.com/linklab-uva/RACECAR_DATA
```
# some missing steps here getting the right message definitions so the db3 can be read, could fill them in if needed 
cd S11/T-SOLO-PYLON
mcap convert T-SOLO-PYLON.db3 T-SOLO-PYLON.mcap
```

[T-SOLO-PYLON_truncated.zip](https://github.com/facontidavide/PlotJuggler/files/13127614/T-SOLO-PYLON_truncated.zip)

```
$ mcap info T-SOLO-PYLON_truncated.mcap
library:   mcap go v1.0.4                                              
profile:   ros2                                                        
messages:  2155                                                        
duration:  999.696818ms                                                
start:     2021-10-23T10:48:20.000173913-07:00 (1635011300.000173913)  
end:       2021-10-23T10:48:20.999870731-07:00 (1635011300.999870731)  
compression:
	zstd: [6/6 chunks] [23.67 MiB/15.75 MiB (33.47%)] [15.76 MiB/sec] 
channels:
	(1)  /vehicle_3/luminar_points                              17 msgs (17.01 Hz)    : sensor_msgs/msg/PointCloud2 [ros2msg]                         
	(2)  /vehicle_3/novatel_bottom/bestpos                      19 msgs (19.01 Hz)    : novatel_oem7_msgs/msg/BESTPOS [ros2msg]                       
	(3)  /vehicle_3/novatel_bottom/bestvel                      21 msgs (21.01 Hz)    : novatel_oem7_msgs/msg/BESTVEL [ros2msg]                       
	(4)  /vehicle_3/novatel_bottom/rawimu                      125 msgs (125.04 Hz)   : novatel_oem7_msgs/msg/RAWIMU [ros2msg]                        
	(5)  /vehicle_3/novatel_top/bestpos                         20 msgs (20.01 Hz)    : novatel_oem7_msgs/msg/BESTPOS [ros2msg]                       
	(6)  /vehicle_3/novatel_top/bestvel                         21 msgs (21.01 Hz)    : novatel_oem7_msgs/msg/BESTVEL [ros2msg]                       
	(7)  /vehicle_3/novatel_top/rawimu                         125 msgs (125.04 Hz)   : novatel_oem7_msgs/msg/RAWIMU [ros2msg]                        
	(8)  /vehicle_3/radar_front/esr_status1                      1 msgs (1.00 Hz)     : delphi_esr_msgs/msg/EsrStatus1 [ros2msg]                      
	(9)  /vehicle_3/radar_front/esr_status2                      1 msgs (1.00 Hz)     : delphi_esr_msgs/msg/EsrStatus2 [ros2msg]                      
	(12) /vehicle_3/radar_front/esr_status5                      7 msgs (7.00 Hz)     : delphi_esr_msgs/msg/EsrStatus5 [ros2msg]                      
	(13) /vehicle_3/radar_front/esr_status7                      7 msgs (7.00 Hz)     : delphi_esr_msgs/msg/EsrStatus7 [ros2msg]                      
	(14) /vehicle_3/radar_front/esr_status8                      6 msgs (6.00 Hz)     : delphi_esr_msgs/msg/EsrStatus8 [ros2msg]                      
	(15) /vehicle_3/radar_front/esr_status9                      6 msgs (6.00 Hz)     : delphi_esr_msgs/msg/EsrStatus9 [ros2msg]                      
	(16) /vehicle_3/radar_front/esr_track                        6 msgs (6.00 Hz)     : delphi_esr_msgs/msg/EsrTrack [ros2msg]                        
	(17) /vehicle_3/radar_front/esr_valid1                       3 msgs (3.00 Hz)     : delphi_esr_msgs/msg/EsrValid1 [ros2msg]                       
	(18) /vehicle_3/radar_front/esr_valid2                       5 msgs (5.00 Hz)     : delphi_esr_msgs/msg/EsrValid2 [ros2msg]                       
	(19) /vehicle_3/radar_front/esr_vehicle1                    28 msgs (28.01 Hz)    : delphi_esr_msgs/msg/EsrVehicle1 [ros2msg]                     
	(20) /vehicle_3/radar_front/esr_vehicle2                    28 msgs (28.01 Hz)    : delphi_esr_msgs/msg/EsrVehicle2 [ros2msg]                     
	(21) /vehicle_3/radar_front/radar_visz_moving                1 msgs (1.00 Hz)     : visualization_msgs/msg/Marker [ros2msg]                       
	(23) /vehicle_3/radar_left/detection                       578 msgs (578.18 Hz)   : delphi_mrr_msgs/msg/Detection [ros2msg]                       
	(24) /vehicle_3/radar_left/header_information_detections    33 msgs (33.01 Hz)    : delphi_mrr_msgs/msg/MrrHeaderInformationDetections [ros2msg]  
	(25) /vehicle_3/radar_left/marker                          579 msgs (579.18 Hz)   : visualization_msgs/msg/Marker [ros2msg]                       
	(26) /vehicle_3/radar_left/vehicle_state                    28 msgs (28.01 Hz)    : delphi_mrr_msgs/msg/VehicleStateMsg2 [ros2msg]                
	(27) /vehicle_3/radar_right/detection                      207 msgs (207.06 Hz)   : delphi_mrr_msgs/msg/Detection [ros2msg]                       
	(28) /vehicle_3/radar_right/header_information_detections   34 msgs (34.01 Hz)    : delphi_mrr_msgs/msg/MrrHeaderInformationDetections [ros2msg]  
	(29) /vehicle_3/radar_right/marker                         202 msgs (202.06 Hz)   : visualization_msgs/msg/Marker [ros2msg]                       
	(30) /vehicle_3/radar_right/vehicle_state                   28 msgs (28.01 Hz)    : delphi_mrr_msgs/msg/VehicleStateMsg2 [ros2msg]                
	(31) /vehicle_3/local_odometry                              19 msgs (19.01 Hz)    : nav_msgs/msg/Odometry [ros2msg]                               
attachments: 0
metadata: 0
```

lucasw, 2023-10-25:
I was using a plotjuggler built with catkin above (and commingled with other ros1 packages), but building with colcon was mostly successful, though there's something wrong with sourcing install/setup.bash after the colcon build:

```
not found: "/home/lucasw/ros/ros2_plotjuggler/install/plotjuggler/share/plotjuggler/local_setup.bash"
```

I can't start plotjuggler via `ros2 run plotjuggler plotjuggler` because of that, but I can launch it where the executable was put into `install/` inside my colcon workspace:

```
./install/plotjuggler/lib/plotjuggler/plotjuggler 
```

And then I can load both ros2 and ros1 mcap bags as desired (this colcon build only has a ros2 subscriber though, so I would have to go back to the catkin built plotjuggler for live ros1 topics).

facontidavide, 2026-09-20:
PlotJuggler 4's MCAP loader now maps the 'cdr' message encoding to the channel's schema encoding (ros2msg/ros1msg) and the ROS parser resolves both, so mcap files produced by `mcap convert` from db3/ros1 bags load correctly. Closing as fixed in PJ4.

---
*PlotJuggler 4 is now the official release and the default branch of this repo; PlotJuggler 3 is end-of-life and no longer receives features or fixes. If this still affects you in PJ4, please reopen or file a fresh issue — it will be looked at.*

closed 2026-09-20 by facontidavide