PlotJuggler/PlotJuggler #881: Creating a ROS2 message using arrays and bounded types gives not enough memory in buffer error

RahulRavichandran1, 2023-11-03:
When a ROS2 message with bounded types is used, for example:
`int32[<=5] up_to_five_integers_array`

Importing a Rosbag with this message, gives an error:

```
The plugin [DataLoad ROS2 bags] thrown the following exception: 

 Not enough memory in the buffer stream
```

Once i change the message to and create a rosbag :
int32[] up_to_five_integers_array

the error message disappears.

Could this is fixed, to allow bounded types array to be imported in plotjuggler?

kdhansen, 2024-06-14:
I'd like to second this. It is also mentioned in issue #836 

### Sponsoring
I'd maybe able to sponsor this fix.

### How to Reproduce
I made a repository [https://github.com/kdhansen/bounded-juggle](https://github.com/kdhansen/bounded-juggle) with a publisher and two message definitions. One with a bounded sequence and one with an unbounded.

1. Clone the repo.
2. Build with colcon `colcon build`
3. `source install/setup.bash`
4. Run the publisher `ros2 run bounded_juggle publish_bounded`
5. Open PlotJuggler and subscribe to both topics `/numbers` and `bounded_numbers`
6. PlotJuggler crashes with output:

> terminate called after throwing an instance of 'eprosima::fastcdr::exception::NotEnoughMemoryException'
>   what():  Not enough memory in the buffer stream
> Stack trace (most recent call last) in thread 55026:
> #24   Object "", at 0xffffffffffffffff, in 
> BFD: DWARF error: section .debug_info is larger than its filesize! (0x93f189 vs 0x531098)
> #23   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f09ccc92352, in clone
> #22   Source "/build/glibc-LcI20x/glibc-2.31/nptl/pthread_create.c", line 477, in start_thread [0x7f09cd0b9608]
> #21   Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f09ccfa5df3, in std::error_code::default_error_condition() const
> #20   Object "/opt/ros/galactic/lib/plotjuggler_ros/libDataStreamROS2.so", at 0x7f09b007e97a, in QPointer<QObject>::~QPointer()
> #19   Object "/opt/ros/galactic/lib/librclcpp.so", at 0x7f09ce60d136, in rclcpp::Executor::spin_once(std::chrono::duration<long, std::ratio<1l, 1000000000l> >)
> #18   Object "/opt/ros/galactic/lib/librclcpp.so", at 0x7f09ce615a00, in rclcpp::Executor::spin_once_impl(std::chrono::duration<long, std::ratio<1l, 1000000000l> >)
> #17   Object "/opt/ros/galactic/lib/librclcpp.so", at 0x7f09ce610fa4, in rclcpp::Executor::execute_any_executable(rclcpp::AnyExecutable&)
> #16   Object "/opt/ros/galactic/lib/librclcpp.so", at 0x7f09ce6106e7, in rclcpp::Executor::execute_subscription(std::shared_ptr<rclcpp::SubscriptionBase>)
> #15   Object "/opt/ros/galactic/lib/librclcpp.so", at 0x7f09ce60ffa3, in rclcpp::Executor::add_callback_groups_from_nodes_associated_to_executor()
> #14   Object "/opt/ros/galactic/lib/librclcpp.so", at 0x7f09ce60f4b7, in rclcpp::Executor::spin_node_some(std::shared_ptr<rclcpp::Node>)
> #13   Object "/opt/ros/galactic/lib/librclcpp.so", at 0x7f09ce621f6a, in rclcpp::GenericSubscription::handle_message(std::shared_ptr<void>&, rclcpp::MessageInfo const&)
> #12   Object "/opt/ros/galactic/lib/plotjuggler_ros/libDataStreamROS2.so", at 0x7f09b00804fa, in DataStreamROS2::messageCallback(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<rclcpp::SerializedMessage>)
> #11   Object "/opt/ros/galactic/lib/plotjuggler_ros/libDataStreamROS2.so", at 0x7f09b008040b, in DataStreamROS2::messageCallback(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<rclcpp::SerializedMessage>)
> #10   Object "/opt/ros/galactic/lib/plotjuggler_ros/libDataStreamROS2.so", at 0x7f09b0097e0e, in PJ::CompositeParser::parseMessage(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, PJ::MessageRef, double&)
> #9    Object "/opt/ros/galactic/lib/plotjuggler_ros/libDataStreamROS2.so", at 0x7f09b009adc8, in IntrospectionParser::parseMessage(PJ::MessageRef, double&)
> #8    Object "/opt/ros/galactic/lib/plotjuggler_ros/libDataStreamROS2.so", at 0x7f09b00c03da, in Ros2Introspection::Parser::deserializeIntoFlatMessage(rcutils_uint8_array_t const*, Ros2Introspection::FlatMessage*) const
> #7    Object "/opt/ros/galactic/lib/plotjuggler_ros/libDataStreamROS2.so", at 0x7f09b00bed4d, in Ros2Introspection::Parser::topicInfo() const
> #6    Object "/opt/ros/galactic/lib/libfastcdr.so.1", at 0x7f09b01ba453, in 
> #5    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f09ccf79698, in __cxa_throw
> #4    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f09ccf793e6, in std::terminate()
> #3    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f09ccf7937b, in std::rethrow_exception(std::__exception_ptr::exception_ptr)
> #2    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f09ccf6d8d0, in __cxa_throw_bad_array_new_length
> #1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f09ccb95858, in abort
> #0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f09ccbb600b, in gsignal
> Aborted (Signal sent by tkill() 54345 1000)

8. Subscribe to just `/numbers` and everything runs fine.

YannickdeHoop, 2025-07-03:
same issue here

facontidavide, 2025-07-03:
correct, I don't support yet that syntax. I didn't even know it existed.

I will look into this