PlotJuggler/PlotJuggler #882: Black screen on start up

FrGe2016, 2023-11-07:
Thanks for contributing to PlotJuggler. You are great!

ROS2 Plotjuggler was working perfectly until today. now i only have a black screen

Describe the problem and the expected behavior. Use screenshots of video if approriate.

# Steps to reproduce (important)
 ros2 run plotjuggler plotjuggler
After the funny image, only a black window with a white header is displayed.
The following message is displayed in the console: 
Loading compatible plugins from directory:  "/opt/ros/humble/lib/plotjuggler_ros"
"libDataLoadROS2.so is a DataLoader plugin"
"libDataStreamROS2.so is a DataStreamer plugin"
[WARN] [1699324843.402412164] [rclcpp]: logging was initialized more than once
"libTopicPublisherROS2.so is a StatePublisher plugin"
Number of plugins loaded:  3 

Loading compatible plugins from directory:  "/opt/ros/humble/lib/plotjuggler"
"libDataLoadCSV.so is a DataLoader plugin"
"libDataLoadMCAP.so is a DataLoader plugin"
"libDataLoadULog.so is a DataLoader plugin"
"libDataStreamSample.so is a DataStreamer plugin"  ...disabled, unless option -t is used
"libDataStreamUDP.so is a DataStreamer plugin"
"libDataStreamWebSocket.so is a DataStreamer plugin"
"libDataStreamZMQ.so is a DataStreamer plugin"
"libParserDataTamer.so is a MessageParser plugin"
"libParserROS1.so is a MessageParser plugin"
"libParserROS2.so is a MessageParser plugin"
"libPublisherCSV.so is a StatePublisher plugin"
Type conversion already registered from type QString to type QwtText
"libToolboxFFT.so is a Toolbox plugin"
"libToolboxLuaEditor.so is a Toolbox plugin"
Type conversion already registered from type QString to type QwtText
"libToolboxQuaternion.so is a Toolbox plugin"
Number of plugins loaded:  13 

Loading compatible plugins from directory:  "/home/francois/.local/share/PlotJuggler"
Number of plugins loaded:  0 

QOpenGLWidget: Failed to create context
QOpenGLWidget: Failed to create context
QOpenGLWidget: Failed to create context
qt.qpa.backingstore: composeAndFlush: QOpenGLContext creation failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed

- Describe your platform / Operative System.
- Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".
- If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.

FrGe2016, 2023-11-07:
Finaly solved after system restart

sverrevr, 2023-12-06:
I have the same problem, it was not solved by a restart. Running plot juggler innside docker if that matters.

facontidavide, 2023-12-06:
disable opengl in Preference on command line. It is the only solution I am aware of

closed 2023-12-06 by facontidavide

sverrevr, 2023-12-06:
Could you elaborate a bit, how do I disable opengl?