PlotJuggler/PlotJuggler #886: Request for flatpak distribution support

SalimTerryLi, 2023-11-25:
# Problem description

Crashed into the exactly same issue as #816, on Mint 21. So I've made a flatpak one to overcome the many problems that AppImage raises: https://github.com/SalimTerryLi/flathub/tree/plotjuggler

May I ask if @facontidavide willing to take over the maintenance of that branch and send it to flathub, or just let me do the redistribution?

Kindly please check if there's any branding or license issue (such as AppID/URI)

facontidavide, 2025-12-23:
I will NOT maintain flatpack. Making sure that snaps / APPIMAGE / ROS / Windows / Mac works is already way too much work.

if you want to maintain it or submit a PR, you are welcome, but I am going to put zero effort into this

facontidavide, 2026-09-20:
The maintainer has stated he won't maintain a Flatpak package given the existing AppImage/snap/ROS/Windows/Mac burden — community-maintained Flatpaks and PRs are welcome, but this isn't an officially supported channel. Closing as wontfix.

---
*PlotJuggler 4 is now the official release and the default branch of this repo; PlotJuggler 3 is end-of-life and no longer receives features or fixes. If this still affects you in PJ4, please reopen or file a fresh issue — it will be looked at.*

closed 2026-09-20 by facontidavide

SammysHP, 2026-09-24:
Flathub, the primary Flatpak repository, does not allow applications containing AI assisted code.