PlotJuggler/PlotJuggler #891: Unable to compile on Ubuntu 22.04.3

BCGelandwagen, 2023-12-06:
# Problem description

I followed the instruction to build 

> git clone --recurse-submodules https://github.com/facontidavide/PlotJuggler.git
cd PlotJuggler
mkdir build; cd build
cmake ..
make -j

but failed at 88%
> Consolidate compiler generated dependencies of target plotjuggler
[ 88%] Linking CXX executable ../bin/plotjuggler
/usr/bin/ld: CMakeFiles/plotjuggler.dir/plotjuggler_autogen/mocs_compilation.cpp.o:(.data.rel.ro._ZTV10PlotDocker[_ZTV10PlotDocker]+0x38): undefined reference to `ads::CDockContainerWidget::event(QEvent*)'
/usr/bin/ld: CMakeFiles/plotjuggler.dir/plot_docker.cpp.o: in function `PlotDocker::plotAt(int)':
/home/domino/PlotJuggler/PlotJuggler/plotjuggler_app/plot_docker.cpp:247: undefined reference to `ads::CDockContainerWidget::dockArea(int) const'
/usr/bin/ld: CMakeFiles/plotjuggler.dir/plot_docker.cpp.o: in function `QtPrivate::QFunctorSlotObject<PlotDocker::PlotDocker(QString, PJ::PlotDataMapRef&, QWidget*)::{lambda()#1}, 0, QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*)':
.
.
.
/usr/bin/ld: ../lib/libqt_advanced_docking.a(mocs_compilation.cpp.o):(.data.rel.ro._ZTVN3ads12CDockManagerE[_ZTVN3ads12CDockManagerE]+0x38): undefined reference to `ads::CDockContainerWidget::event(QEvent*)'
collect2: error: ld returned 1 exit status
make[2]: *** [plotjuggler_app/CMakeFiles/plotjuggler.dir/build.make:1166: bin/plotjuggler] Error 1
make[1]: *** [CMakeFiles/Makefile2:1233: plotjuggler_app/CMakeFiles/plotjuggler.dir/all] Error 2
make: *** [Makefile:156: all] Error 2

# Steps to reproduce (important)
Linux Ubuntu 22.04.3 LTS 
x64 i7-6700T 
x11 windowing system

Am I missing any dependencies? 

I can install PlotJuggler using snap and it runs fine but I want to add the CAN plug-in, which requires me to do full compilation and installation of PJ.

Thank you,
BC

BCGelandwagen, 2023-12-06:
Downloaded the appimage from plotjuggler-CAN-dbs with plugin built and can run the app now. Great tool. Thank you.

facontidavide, 2023-12-06:
I still don't understand what the problem was...

BCGelandwagen, 2023-12-06:
I believe I was missing some of the dependencies in the COMPILE.md steps. 

After grabbing all the dependencies and following the steps:
> git clone --recurse-submodules https://github.com/facontidavide/PlotJuggler.git
cd PlotJuggler
mkdir build; cd build
cmake ..
make -j
sudo make install

I was able to install PlogJugger system wide

> Install the project...
-- Install configuration: "RelWithDebInfo"
-- Up-to-date: /usr/local/share/applications/PlotJuggler.desktop
-- Up-to-date: /usr/local/share/icons/plotjuggler.svg
-- Up-to-date: /usr/local/lib/libplotjuggler_base.a
-- Up-to-date: /usr/local/include
-- Up-to-date: /usr/local/include/PlotJuggler
-- Up-to-date: /usr/local/include/PlotJuggler/plotdatabase.h
-- Up-to-date: /usr/local/include/PlotJuggler/plotdata.h
-- Up-to-date: /usr/local/include/PlotJuggler/datastreamer_base.h
-- Up-to-date: /usr/local/include/PlotJuggler/alphanum.hpp
-- Up-to-date: /usr/local/include/PlotJuggler/string_ref_sso.h
-- Up-to-date: /usr/local/include/PlotJuggler/toolbox_base.h
-- Up-to-date: /usr/local/include/PlotJuggler/dataloader_base.h
-- Up-to-date: /usr/local/include/PlotJuggler/reactive_function.h
-- Up-to-date: /usr/local/include/PlotJuggler/pj_plugin.h
-- Up-to-date: /usr/local/include/PlotJuggler/plotwidget_base.h
-- Up-to-date: /usr/local/include/PlotJuggler/special_messages.h
-- Up-to-date: /usr/local/include/PlotJuggler/statepublisher_base.h
-- Up-to-date: /usr/local/include/PlotJuggler/timeseries.h
-- Up-to-date: /usr/local/include/PlotJuggler/ring_span.hpp
-- Up-to-date: /usr/local/include/PlotJuggler/util
-- Up-to-date: /usr/local/include/PlotJuggler/util/delayed_callback.hpp
-- Up-to-date: /usr/local/include/PlotJuggler/stringseries.h
-- Up-to-date: /usr/local/include/PlotJuggler/transform_function.h
-- Up-to-date: /usr/local/include/PlotJuggler/fmt
-- Up-to-date: /usr/local/include/PlotJuggler/fmt/core.h
-- Up-to-date: /usr/local/include/PlotJuggler/fmt/format.h
-- Up-to-date: /usr/local/include/PlotJuggler/fmt/locale.h
-- Up-to-date: /usr/local/include/PlotJuggler/fmt/args.h
-- Up-to-date: /usr/local/include/PlotJuggler/fmt/xchar.h
-- Up-to-date: /usr/local/include/PlotJuggler/fmt/color.h
-- Up-to-date: /usr/local/include/PlotJuggler/fmt/ostream.h
-- Up-to-date: /usr/local/include/PlotJuggler/fmt/chrono.h
-- Up-to-date: /usr/local/include/PlotJuggler/fmt/compile.h
-- Up-to-date: /usr/local/include/PlotJuggler/fmt/format-inl.h
-- Up-to-date: /usr/local/include/PlotJuggler/fmt/os.h
-- Up-to-date: /usr/local/include/PlotJuggler/fmt/printf.h
-- Up-to-date: /usr/local/include/PlotJuggler/fmt/ranges.h
-- Up-to-date: /usr/local/include/PlotJuggler/svg_util.h
-- Up-to-date: /usr/local/include/PlotJuggler/messageparser_base.h
-- Up-to-date: /usr/local/lib/cmake/plotjuggler/plotjugglerTargets.cmake
-- Up-to-date: /usr/local/lib/cmake/plotjuggler/plotjugglerTargets-relwithdebinfo.cmake
-- Up-to-date: /usr/local/lib/cmake/plotjuggler/plotjugglerConfig.cmake
-- Up-to-date: /usr/local/lib/cmake/plotjuggler/plotjugglerConfigVersion.cmake
-- Up-to-date: /usr/local/lib/libplotjuggler_qwt.a
-- Up-to-date: /usr/local/bin/plotjuggler
-- Up-to-date: /usr/local/bin/libDataLoadCSV.so
-- Up-to-date: /usr/local/bin/libDataLoadMCAP.so
-- Up-to-date: /usr/local/bin/libDataLoadULog.so
-- Up-to-date: /usr/local/bin/libDataStreamSample.so
-- Up-to-date: /usr/local/bin/libDataStreamWebSocket.so
-- Up-to-date: /usr/local/bin/libDataStreamUDP.so
-- Up-to-date: /usr/local/bin/libDataStreamZMQ.so
-- Up-to-date: /usr/local/bin/libPublisherCSV.so
-- Up-to-date: /usr/local/bin/libToolboxQuaternion.so
-- Up-to-date: /usr/local/bin/libToolboxFFT.so
-- Up-to-date: /usr/local/bin/libToolboxLuaEditor.so
-- Up-to-date: /usr/local/bin/libProtobufParser.so
-- Installing: /usr/local/bin/libParserROS1.so
-- Installing: /usr/local/bin/libParserROS2.so
-- Up-to-date: /usr/local/bin/libParserDataTamer.so

domino@domino:~/PlotJuggler/build$

closed 2023-12-06 by BCGelandwagen