PlotJuggler/PlotJuggler #894: Question Reactive Script

fazerlab, 2023-12-15:
I´m trying Reactive Script and have questions:

I startedwith a basic script:

```
file = io.open("testwrite.txt","w")
myText = "Hello"
file:write(myText)
file:close()
```
[image]

Raise Error:

[image]

This script was tested with Lua 54 and work fine.

Another question:  Is possible install Lua modules? for example LuaSocket https://github.com/lunarmodules/luasocket

My setup:

Win10
PlotJuggler-Windows-3.6.0

Thank you

simonsag96, 2024-01-27:
[image]

Here are the lua libraries loaded for the reactive function editor.
I assume your script isn't working bc the io library is not explicitly loaded.

Adding the library could be done there, but I imagine that especially file io would be pretty slow for doing on cursor movement.

facontidavide, 2026-09-20:
Answering the two questions, and closing as this predates PlotJuggler 4.

@simonsag96 was right: the `io` library was not exposed to the script sandbox, which is why `io.open` failed. That is still deliberate in PJ4 — data processors and scripts are sandboxed to computation, with no filesystem or network access, and the Python processor allows only an explicit list of imports. So arbitrary Lua modules such as LuaSocket are not loadable either.

If you need to get data out of PlotJuggler, use a data-export toolbox or a streaming source rather than script-side file I/O.

---
*PlotJuggler 4 is now the official release and the default branch of this repo; PlotJuggler 3 is end-of-life and no longer receives features or fixes. If this still affects you in PJ4, please reopen or file a fresh issue — it will be looked at.*

closed 2026-09-20 by facontidavide