PlotJuggler/PlotJuggler #896: Loading multiple rosbags becomes progressively slower

parakhm95, 2023-12-30:
When I am loading multiple bag files in Plotjuggler for analysis, if they are over 2 gigabyte each, the loading of the bags becomes progressively slower. From a programming point of view, it feels like it grows exponentially and might be an issue of how the storage data type is expanded to accommodate more data and how it is maybe copied over to expand instead of pre-allocating memory for it. 

I am using Ubuntu 20.04 and loading experimental data that I cannot share with you but problem should be reproducible with large bag files.

facontidavide, 2023-12-30:
When you load them, do you delete or keep the previous data?

machinemg, 2024-01-02:
I do this by launching plotjuggler from the command line and then selecting multiple files at once. When asked to delete the previous data, I select No but if I remember it doesn't ask me that if I am loading multiple files through selection in file explorer menu.

parakhm95, 2024-01-02:
I do this by launching plotjuggler from the command line and then selecting multiple files at once. When asked to delete the previous data, I select No but if I remember it doesn't ask me that if I am loading multiple files through selection in file explorer menu.

facontidavide, 2024-01-02:
when loading data from file, ALL your data is stored in memory. Maybe you are just running out of memory.

Anyway, once I have time (not tight now), I should investigate if anything can be done.

parakhm95, 2024-01-02:
Okay so I just finished running a test on my laptop with 32 GB machine. I copied a 3.05 GB bag thrice into `/tmp` and started loading through plotjuggler by selecting all of them at once. My RAM at the beginning of this operation was used close to ~ 5 GB out of 32 GB. 

First loading operation took 77.5 seconds, second one took 96.5 seconds, and third one took 119.4 seconds. This is the exact issue I am referring to. I timed this externally as well in addition to the terminal reporting the same values from plotjuggler's printouts. My RAM usage at the end of the operation was about 25 GB out of 32 GB available. The swap didn't change from the beginning to the end of this operation and stayed at ~300 MB at all times.

machinemg, 2024-01-02:
I have sent you the bag to your email ID mentioned on your github profile. You just need to copy it thrice and then you can try out the experiment on your own. 

Let me know if you face issues with reproducing the error.

facontidavide, 2024-01-03:
I think I understand what the problem is, but there is something I need to understand first about your use case

> When I am loading multiple bag files in Plotjuggler for analysis ...

I am not sure what you mean here.

1. You want to load one file and LATER you want to open another file. is it OK if all the data from the first file is deleted
2. You want to compare two bags, side by side
3. Something else

**CASE 1:**

This is an easy solution. I can add a "clean all" implicitly when loading the rosbag

**CASE 2:**

You must select multiple files in the FileLoader at once. This will add a prefix and effectively prevent PlotJuggler from mergind data into the same timeseries.

parakhm95, 2024-01-03:
For me, it is the case 2, I am loading 5-6 rosbags together to compare results and plots between many runs of my experiments.

parakhm95, 2024-02-19:
Reverified loading times on v3.9.0 with the same file to be 50 s, 56 s, and 64 s, respectively. I will consider this resolved.

facontidavide, 2024-02-19:
I did some investigation, but I was not able to find anything (with more time, maybe I could, but... I don't have much time!).

Happy that this is not a blocker

parakhm95, 2024-02-19:
Lol, I always thought that you added the `add prefix and merge` button to resolve this issue. 😄

facontidavide, 2024-02-19:
no, that is something I did selfishly for myself (I also need to compare datasets)