PlotJuggler/PlotJuggler #897: Compilation problem on Macbook air M1

ibeyzaekici, 2023-12-31:
Hello, I am encountering an issue while attempting to configure and build PlotJuggler. The error is related to the C compiler, and it seems to be failing during a simple test program compilation. The build process is encountering a "file cannot be mmap()ed" error with a specific path related to the Qt library.
1. *Platform / Operating System:** macOS Sonoma 14.2.1
2. **PlotJuggler Version: 3.8.4
3. **CMake Version: 3.28.1
4. **Xcode Version: Xcode 15.1

This is the error i am getting: 
[image]
[image]

I would be appreciated if you help me on that. Thank you.

NothanUmber, 2024-02-09:
Same problem here, if somebody finds a solution, mentioning it here would be great!

ubaldot, 2024-02-29:
Maybe this helps?

https://stackoverflow.com/questions/53633705/cmake-the-c-compiler-is-not-able-to-compile-a-simple-test-program

ubaldot, 2024-03-07:
I had the same problem and I tried the solution that I linked tonight and it didn't work.
That is, I opened the  `./src/PlotJuggler/CMakelist.txt ` file and add I added the following: 

`set(CMAKE_TRY_COMPILE_TARGET_TYPE "STATIC_LIBRARY")`

just before `PROJECT.` (it is at the very beginning). 
Then, try to build again. It failed at 75% of build in the same linking error:

`ld: file cannot be mmap()ed, errno=22 path=/opt/homebrew/opt/qt@5/lib in '/opt/homebrew/opt/qt@5/lib'`

ubaldot, 2024-03-07:
**SOLVED**
1. Add the following to your `CMakeList.txt `file:

```
# Include PkgConfig module
find_package(PkgConfig REQUIRED)

# Use pkg_check_modules to find ZeroMQ
pkg_check_modules(ZEROMQ REQUIRED IMPORTED_TARGET libzmq)

# Include ZeroMQ directories
include_directories(${ZEROMQ_INCLUDE_DIRS})

```
2. Don't export the LDFLAG environment variable i.e. use the following:

```
QT_HOME=$(brew --prefix qt@5)
export CPPFLAGS="-I $QT_HOME/include"
export PKG_CONFIG_PATH="$QT_HOME/lib/pkgconfig"
```

instead of the following (as written in the guide):

```
QT_HOME=$(brew --prefix qt@5)
export CPPFLAGS="-I $QT_HOME/include"
export PKG_CONFIG_PATH="$QT_HOME/lib/pkgconfig"
export LDFLAGS="$QT_HOME/lib"
```

rizvanovic, 2024-04-25:
How do you even get it to compile on an M1 macbook. For me it complains about the arm64 arch.

ubaldot, 2024-04-25:
> How do you even get it to compile on an M1 macbook. For me it complains about the arm64 arch.

Here are the files I changed:

https://github.com/facontidavide/PlotJuggler/pull/950/files

what error message you get extactly?

closed 2025-12-07 by facontidavide