PlotJuggler/PlotJuggler #901: The plugin [DataLoad ROS bags] thrown the following exception:    map::at

Zheqiu2023, 2024-01-02:
Thanks for contributing to PlotJuggler. It's useful and the start animation is funny!

I encountered a problem when trying to load data from a rosbag.
when I run `rosrun plotjuggler plotjuggler`,  the following problem appears:

> "libParserROS1.so" :  "Cannot load library /opt/ros/noetic/lib/plotjuggler/libParserROS1.so: (librosx_introspection.so: cannot open shared object file: No such file or directory)"
>"libParserROS2.so" :  "Cannot load library /opt/ros/noetic/lib/plotjuggler/libParserROS2.so: (librosx_introspection.so: cannot open shared object file: No such file or directory)"

 After opening a rosbag and selecting the topic the following exception occurs:

[image]
[image]

I am using plotjuggler 3.8.4 with ros-noetic on ubuntu-20.04
The used rosbag is the following:
[test.zip](https://github.com/facontidavide/PlotJuggler/files/13811367/test.zip)

Any ideas how this can be solved?Thanks a lot!

jamesdarrenmuir, 2024-01-03:
I have the same issue (and a crash if I try to start the ROS Topic Subscriber) when installing the latest version using `apt`, but not when building from source following the instructions [here](/home/jamesmuir/ws_plotjuggler/devel/.private/plotjuggler/lib/plotjuggler).

Results when launching version from `apt`:

```
jamesmuir@james-laptop:~$ apt show ros-noetic-plotjuggler-ros
Package: ros-noetic-plotjuggler-ros
Version: 2.0.0-1focal.20231221.172804
...
jamesmuir@james-laptop:~$ apt show ros-noetic-plotjuggler
Package: ros-noetic-plotjuggler
Version: 3.8.4-1focal.20231221.171007
...
jamesmuir@james-laptop:~$ rosrun plotjuggler plotjuggler 
Loading compatible plugins from directory:  "/opt/ros/noetic/lib/plotjuggler"
...
"libParserROS1.so" :  "Cannot load library /opt/ros/noetic/lib/plotjuggler/libParserROS1.so: (librosx_introspection.so: cannot open shared object file: No such file or directory)"
"libParserROS2.so" :  "Cannot load library /opt/ros/noetic/lib/plotjuggler/libParserROS2.so: (librosx_introspection.so: cannot open shared object file: No such file or directory)"
```

Results when launching version installed from source:

```
jamesmuir@james-laptop:~/ws_plotjuggler$ rosrun plotjuggler plotjuggler 
Loading compatible plugins from directory:  "/home/jamesmuir/ws_plotjuggler/devel/.private/plotjuggler/lib/plotjuggler"
...
"libParserROS1.so is a MessageParser plugin"
"libParserROS2.so is a MessageParser plugin"
...
"librosx_introspection.so" :  "Failed to extract plugin meta data from '/home/jamesmuir/ws_plotjuggler/devel/.private/plotjuggler/lib/plotjuggler/librosx_introspection.so'"
```

<details>
<summary>
Full error message from crash when trying to start the ROS Topic Subscriber using the version from `apt`:
</summary>

```
jamesmuir@james-laptop:~$ rosrun plotjuggler plotjuggler 
Loading compatible plugins from directory:  "/opt/ros/noetic/lib/plotjuggler"
"libDataLoadCSV.so is a DataLoader plugin"
"libDataLoadMCAP.so is a DataLoader plugin"
"libDataLoadULog.so is a DataLoader plugin"
"libDataStreamSample.so is a DataStreamer plugin"  ...disabled, unless option -t is used
"libDataStreamUDP.so is a DataStreamer plugin"
"libDataStreamWebSocket.so is a DataStreamer plugin"
"libDataStreamZMQ.so is a DataStreamer plugin"
"libParserDataTamer.so is a MessageParser plugin"
"libParserROS1.so" :  "Cannot load library /opt/ros/noetic/lib/plotjuggler/libParserROS1.so: (librosx_introspection.so: cannot open shared object file: No such file or directory)"
"libParserROS2.so" :  "Cannot load library /opt/ros/noetic/lib/plotjuggler/libParserROS2.so: (librosx_introspection.so: cannot open shared object file: No such file or directory)"
"libProtobufParser.so is a MessageParser plugin"
"libPublisherCSV.so is a StatePublisher plugin"
Type conversion already registered from type QString to type QwtText
"libToolboxFFT.so is a Toolbox plugin"
"libToolboxLuaEditor.so is a Toolbox plugin"
Type conversion already registered from type QString to type QwtText
"libToolboxQuaternion.so is a Toolbox plugin"
Number of plugins loaded:  12 

Loading compatible plugins from directory:  "/opt/ros/noetic/lib/plotjuggler_ros"
"libDataLoadROS.so is a DataLoader plugin"
"libDataStreamROS.so is a DataStreamer plugin"
"libRosoutPublisher.so is a StatePublisher plugin"
"libRosTopicPublisher.so is a StatePublisher plugin"
Number of plugins loaded:  4 

Loading compatible plugins from directory:  "/home/jamesmuir/catkin_workspaces/overlay_ws/devel/lib/plotjuggler_ros"
Number of plugins loaded:  0 

Loading compatible plugins from directory:  "/home/jamesmuir/catkin_workspaces/base_ws/devel/lib/plotjuggler_ros"
[... 26 lines cut ...]
#19   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f7280eb228b, in QMetaObject::activate(QObject*, int, int, void**)
#18   Object "/opt/ros/noetic/lib/plotjuggler/plotjuggler", at 0x556ba9af0b89, in 
#17   Object "/opt/ros/noetic/lib/plotjuggler/plotjuggler", at 0x556ba9aeef23, in 
#16   Object "/opt/ros/noetic/lib/plotjuggler/plotjuggler", at 0x556ba9b37d74, in boost::system::system_error::what() const
#15   Object "/opt/ros/noetic/lib/plotjuggler/plotjuggler", at 0x556ba9b2be72, in boost::system::system_error::what() const
#14   Object "/opt/ros/noetic/lib/plotjuggler_ros/libDataStreamROS.so", at 0x7f725867ae21, in DataStreamROS::start(QStringList*)
#13   Object "/opt/ros/noetic/lib/plotjuggler_ros/libDataStreamROS.so", at 0x7f7258679a47, in DataStreamROS::extractInitialSamples()
#12   Object "/home/jamesmuir/catkin_workspaces/base_ws/devel/lib/libroscpp.so", at 0x7f72822514b2, in ros::CallbackQueue::callAvailable(ros::WallDuration)
#11   Object "/home/jamesmuir/catkin_workspaces/base_ws/devel/lib/libroscpp.so", at 0x7f728224fe81, in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*)
#10   Object "/home/jamesmuir/catkin_workspaces/base_ws/devel/lib/libroscpp.so", at 0x7f728229fe2f, in ros::SubscriptionQueue::call()
#9    Object "/opt/ros/noetic/lib/plotjuggler_ros/libDataStreamROS.so", at 0x7f7258691bd8, in ros::SubscriptionCallbackHelperT<boost::shared_ptr<RosIntrospection::ShapeShifter const> const&, void>::call(ros::SubscriptionCallbackHelperCallParams&)
#8    Object "/opt/ros/noetic/lib/plotjuggler_ros/libDataStreamROS.so", at 0x7f7258683bb5, in boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<RosIntrospection::ShapeShifter const> const&)>, void, boost::shared_ptr<RosIntrospection::ShapeShifter const> >::invoke(boost::detail::function::fu [... cut ...]
#3    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f7280ac837b, in std::rethrow_exception(std::__exception_ptr::exception_ptr)
#2    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f7280abc8d0, in __cxa_throw_bad_array_new_length
BFD: DWARF error: section .debug_info is larger than its filesize! (0x93f189 vs 0x530f70)
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f72806ff858, in abort
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f728072000b, in gsignal
Aborted (Signal sent by tkill() 82056 1000)
Aborted (core dumped)
```
</details>

Zheqiu2023, 2024-01-03:
It looks very similar to my problem. The link doesn't work. Could you please give me a new one? thanks

facontidavide, 2024-01-03:
Ok, I think I know what the issue is. This change should fix it, but you will need to wait for the release of 3.8.5

https://github.com/facontidavide/PlotJuggler/commit/1f802e851fbb04db976e5023e3868207bab69ee6

closed 2024-01-03 by facontidavide