PlotJuggler/PlotJuggler #903: Floating point truncated exception when loading bag with messages with INT64 fields with values bigger than std::numerical_limits<double>::max

mme2lr-bosch, 2024-01-03:
# Problem description

When loading a bag which contains messages with INT64 fields whose values are bigger than `std::numerical_limits<double>::max`, a "Floating point truncated" execption is thrown:

# Steps to reproduce (important)

- OS Ubuntu 22.04
- Plotjuggler 3.8.3
- plotjuggler_msgs 0.2.2
- plotjuggler-ros-plugins 3.8.3
- Reproduction
  - Create bag with message which has INT64 field with value bigger than `std::numerical_limits<double>::max`
  - Open plotjuggler . 
  - File -> Data -> Load the bag

# Root Cause Analysis

When loading the bag, the function member function `convert()` is called for a `Variant` object which contains the too big `INT64` value in `_storage.raw_data[0]`. In [LOC 359](https://github.com/facontidavide/PlotJuggler/blob/main/plotjuggler_plugins/ParserROS/rosx_introspection/include/rosx_introspection/variant.hpp#L359), this value is converted to double using the `convert_impl` function with template arguments `SRC` equal to `int64_t` and `DST` as `double`. 

This function is implemented in [converstion_impl.hpp](https://github.com/facontidavide/PlotJuggler/blob/main/plotjuggler_plugins/ParserROS/rosx_introspection/include/rosx_introspection/details/conversion_impl.hpp#L320-L338) and checks if the conversion can be done exactly and throws a RangeException otherwise in [LOC 334](https://github.com/facontidavide/PlotJuggler/blob/main/plotjuggler_plugins/ParserROS/rosx_introspection/include/rosx_introspection/details/conversion_impl.hpp#L337)

Since the max value of INT64 is bigger than the max value of double, this can happen for messages with INT64 field with too big values.

Problem **cannot** be reproduced with dummy data

Will share bag to reproduce problem soon (currently having issues with uploading bag to github)

facontidavide, 2024-01-09:
PlotJuggler can only manage numbers that can be represented as double.

What would be the desired behavior in this case? Truncation?
There is no other solution

mme2lr-bosch, 2024-01-09:
Ok.

I think a pragmatic workaround would be to ignore the series which contain such values. This would at least allow us to visualize the rest of the bag.

facontidavide, 2024-01-09:
that will take me considerable more time than simply remove the exception.
I will do it once I have time. Feel free of course to create a PR to fix this

mme2lr-bosch, 2024-01-09:
Ok, thanks. I assume simply removing the exception is not something you would merge to master, right?

facontidavide, 2024-01-09:
I guess I might, since I received this complaint more time than I can count.

facontidavide, 2026-09-20:
PlotJuggler 4's ROS parser stores integer fields natively (int64/uint64) via ValueRef instead of forcing every value through double, so large INT64 values no longer trigger the "Floating point truncated" exception. Closing as fixed in PJ4.

---
*PlotJuggler 4 is now the official release and the default branch of this repo; PlotJuggler 3 is end-of-life and no longer receives features or fixes. If this still affects you in PJ4, please reopen or file a fresh issue — it will be looked at.*

closed 2026-09-20 by facontidavide