PlotJuggler/PlotJuggler #904: Plotjuggler does not show correct values with ROS1

NR-Daichi, 2024-01-03:
# Problem description
When plotjuggler is subscribing odometry topic with ROS1, the values in pose/pose/position and pose/pose/orientation are very huge values or 0
For the details, please watch the attached video

Joint_states topics are fine, Plotjuggler displays the right values

# Steps to reproduce (important)
Platform: Ubuntu 20.04, ROS1

Step 1:
install ros plugins by following the instructions written on this page ("How to compile PlotJuggler from source")
https://github.com/PlotJuggler/plotjuggler-ros-plugins

Step 2:
launch plotjuggler and subscribe odometry topic

If anyone know the way to run plotjuggler with ros1 on ubuntu 22.04, please let me know

lucasw, 2024-01-04:
Can you zip up a short rosbag of your odom topic and attach it here?

peci1, 2024-01-04:
I get the same bug with apt-installed plotjuggler on Ubuntu 20.04/Noetic.

Sample odom topic values that get displayed as the nonsense:

```
x: 82.09575053694694
y: -8.41266412125995
z: -0.08068917743157186
```

peci1, 2024-01-04:
Combination of PJ 3.8.5 from testing and PJ-ROS 1.7.0 from snapshot 2023-11-15 works.

sunilsulania, 2024-01-09:
It started happening recently with me as well on ROS1 on ubuntu 20.04 apt installed version. I may have done an apt upgrade, I dont remember. 
Interestingly, Issue is with nav_msgs/Odometry position values. like mentioned in above image.

NR-Daichi, 2024-01-09:
> Can you zip up a short rosbag of your odom topic and attach it here?

[ros_bagfiles.zip](https://github.com/facontidavide/PlotJuggler/files/13871499/ros_bagfiles.zip)
Please have a look at the odometry topic (ex. /right_foot_BF_pin/pose/pose/position/x ) 
I could plot them with rqt_plot
[image]

NR-Daichi, 2024-01-09:
> It started happening recently with me as well on ROS1 on ubuntu 20.04 apt installed version. I may have done an apt upgrade, I dont remember. Interestingly, Issue is with nav_msgs/Odometry position values. like mentioned in above image.

This suddenly happen to me as well. In my case, the Odometry/Twist / Twist / linear is also not correct

facontidavide, 2024-01-09:
Fixed: 
[image]

sunilsulania, 2024-01-09:
Thank you Davide.

Veng97, 2024-01-13:
Can confirm this issue also happens in ROS2.

[image]
[image]

The values shown in plotjuggler does not match the topic echo.

facontidavide, 2024-01-13:
Problem fixed in 3.8.6

Unless you are using that version, don't duplicate this issue

closed 2024-01-16 by facontidavide