PlotJuggler/PlotJuggler #909: Need an option for using PlotJuggler without memes

rccarlson, 2024-01-10:
I was hoping to use PlotJuggler in a professional context to display data for customers, but a meme showing up when we start/restart PlotJuggler would kind of ruin any sense of professionalism we are trying to maintain.

Could a compile flag or fork be added to remove/hide memes?

facontidavide, 2024-01-11:
It is already there. Check --help

facontidavide, 2024-01-11:
Also, I will be happy to provide a version without memes for 999 USD

closed 2024-01-11 by facontidavide

pifehub, 2024-08-30:
Hobby/non-professional user here.

I think having a GUI option to enable/disable memes would be good. (A checkbox in the settings window, or a dialog the very first time Plotjuggler is opened) I can understand people who like them, I can understand people who dislike them. They are certainly unusual. When I tried out Plotjuggler for the first time, the first thing I was greeted with was a meme, which gave me immediate concerns about the credibility of the software / website and I disliked the experience before even using the software. (I was reminded of the days when some software would install browser toolbars without permission).

Another point besides the "professionalism" is distraction. When I analyze data, I want to be focused on analyzing the data. I don't want Plotjuggler to display memes. I don't want the Python interpreter to play funny Tiktok Videos. I don't want a hammer to play a funny song every time I use it. I want a clean tool that is designed to do the task it was made for. Of course I can only speak for myself and not for all users, but I personally prefer "I need to understand this data, let's open Plotjuggler. Ah, yes, that's where the problem comes from, thanks Plotjuggler!" to "I need to understand this data, let's open Plotjuggler. lol, a meme starring Yoda. Man, I really miss Yoda. I wonder if someday we'll get another good Star Wars movie. Let's check IMDB for updates." 

Also, until now I thought the memes exist to make the loading time at the program startup less boring, but I just tried the `-n` option and realized there is no loading time... The memes just exist to delay the program startup so you have 5 seconds to look at a meme.

facontidavide, 2024-08-30:
You have found my deepest secret.
Just use "-n".

Do you want a checkbox in settings? Send a PR.

> When I analyze data, I want to be focused on analyzing the data. I don't want Plotjuggler to display memes

Plotjugger is free software, use it if you find it useful, modify as you want.

facontidavide, 2024-08-30:
> When I tried out Plotjuggler for the first time, the first thing I was greeted with was a meme, which gave me immediate concerns about the credibility of the software / website and I disliked the experience before even using the software. 

Remember that while this is free software and no one pays for its maintenance, I don't need to make everyone happy, nor it makes any difference, from my point of view, if you use it or not 😉