PlotJuggler/PlotJuggler #910: Topic /topic_name: map::at

virajawate, 2024-01-12:
Thanks for contributing to PlotJuggler. You are great!

# Problem description

This issue is coming in recent time Dec23 - Jan 2024. I am not sure about the new patch of any thing
but the issue is - ROS data_types llike nav_msgs::Odometer or std_msgs::MultiArrayFloat32 are not getting 
visualised. The error statement is mentioned in the title, i am getting a pop-up box in the plotjuggler.

# Steps to reproduce (important)

- Describe your platform / Operative System.
- I am using a linux system Ubuntu20 with ROS Noetic installed in it. 
- Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples". -> No
- If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem. -> Yes

facontidavide, 2024-01-16:
if you give me a sample rosbag, I can fix this today

virajawate, 2024-01-16:
Its is company data, but i will definitely get back to you after i have a word with my seniors.

facontidavide, 2024-01-16:
you can share it privately using my email davide.faconti@gmail.com

virajawate, 2024-02-01:
I updated the package now it's working 
Thanks

closed 2024-02-01 by facontidavide