PlotJuggler/PlotJuggler #911: Crash when dragging data into plot area on MacOS

ztqwerty, 2024-01-12:
Thanks for contributing to PlotJuggler. You are great!

I encountered some issues when installing / using PlotJuggler. The program crashed when I try to drag a data into the plot area
OS: MacOS Big Sur 11.7.6

# Steps to reproduce (important)
I compiled PlotJuggler from source on MacOS (Big Sur 11.7.6) with CMake (3.26.0). The compilation process is successful. 

And I can successfully run it after I set environmental variable SYSTEM_VERSION_COMPAT=0 otherwise it reports an error says "Sorry, "plotjuggler" cannot be run on this version of macOS. Qt requires macOS 11.0.0 or later, you have macOS 10.16.0."

Then I am able to load some data (.ulg) from the main UI. The data list can be shown on the left panel. But when I try to drag some data to the plot area, it suddenly crashed. (100% repeatable, no matter I use my own data input or dummy sample data)

The terminal prints logs like below:
qt.qpa.fonts: Populating font family aliases took 730 ms. Replace uses of missing font family "Ubuntu" with one that exists to avoid this cost.
Loading compatible plugins from directory:  "/usr/local/bin"
"libauto_schedule.dylib" :  "'/usr/local/bin/libauto_schedule.dylib' is not a Qt plugin"
"libDataLoadCSV.dylib is a DataLoader plugin"
"libDataLoadMCAP.dylib is a DataLoader plugin"
"libDataLoadULog.dylib is a DataLoader plugin"
"libDataStreamSample.dylib is a DataStreamer plugin"  ...disabled, unless option -t is used
"libDataStreamUDP.dylib is a DataStreamer plugin"
"libDataStreamWebSocket.dylib is a DataStreamer plugin"
"libDataStreamZMQ.dylib is a DataStreamer plugin"
"libHalide.dylib" :  "'/usr/local/bin/libHalide.dylib' is not a Qt plugin"
"libParserDataTamer.dylib is a MessageParser plugin"
"libParserROS1.dylib is a MessageParser plugin"
"libParserROS2.dylib is a MessageParser plugin"
"libPublisherCSV.dylib is a StatePublisher plugin"
Type conversion already registered from type QString to type QwtText
"libToolboxFFT.dylib is a Toolbox plugin"
"libToolboxLuaEditor.dylib is a Toolbox plugin"
Type conversion already registered from type QString to type QwtText
"libToolboxQuaternion.dylib is a Toolbox plugin"
Number of plugins loaded:  13

Loading compatible plugins from directory:  "/Users/Someuser/Library/Application Support/PlotJuggler"
Number of plugins loaded:  0

Could not create pixmap from :/style_light/transparent.png
Could not create pixmap from :/style_light/transparent.png
Could not create pixmap from :/style_light/transparent.png
Could not create pixmap from :/style_light/transparent.png
Could not create pixmap from :/style_light/transparent.png
Could not create pixmap from :/style_light/transparent.png
Stack trace (most recent call last):
#31   Object "QtGui", at 0x10f33e39b, in QDragManager::drag(QDrag*) + 243
#30   Object "libqcocoa.dylib", at 0x112eef3ae, in qt_plugin_instance + 206261
#29   Object "AppKit                              0x00007fff233610b1", at 0x7fff233610b1, in dragImage:at:offset:event:pasteboard:source:slideBack:] + 133
#28   Object "AppKit                              0x00007fff23361bc1", at 0x7fff23361bc1, in dragImage:fromWindow:at:offset:event:pasteboard:source:slideBack:] + 2816
#27   Object "AppKit                              0x00007fff233630e7", at 0x7fff233630e7, in _dragUntilMouseUp:accepted:] + 1213
#26   Object "HIServices", at 0x7fff25aaf4d3, in CoreDragStartDragging + 599
#25   Object "HIServices", at 0x7fff25ab07ed, in DragInApplication + 740
#24   Object "HIServices", at 0x7fff25ab1c60, in DoDropMessage + 41
#23   Object "HIServices", at 0x7fff25ab1e6a, in DoMultipartDropMessage + 104
#22   Object "HIServices", at 0x7fff25ab7bba, in CallReceiveMessageCollectionWithMessage + 102
#21   Object "AppKit", at 0x7fff23365b23, in NSCoreDragReceiveMessageProc + 475
#20   Object "libqcocoa.dylib", at 0x112edbca4, in qt_plugin_instance + 126635
#19   Object "QtGui", at 0x10f30449e, in QWindowSystemInterface::handleDrop(QWindow*, QMimeData const*, QPoint const&, QFlags<Qt::DropAction>, QFlags<Qt::MouseButton>, QFlags<Qt::KeyboardModifier>) + 66
#18   Object "QtGui", at 0x10f3166f7, in QGuiApplicationPrivate::processDrop(QWindow*, QMimeData const*, QPoint const&, QFlags<Qt::DropAction>, QFlags<Qt::MouseButton>, QFlags<Qt::KeyboardModifier>) + 131
#17   Object "QtCore", at 0x10f7ee40d, in QCoreApplication::notifyInternal2(QObject*, QEvent*) + 137
#16   Object "QtWidgets", at 0x10ee927f1, in QApplication::notify(QObject*, QEvent*) + 487
#15   Object "QtWidgets", at 0x10ee91995, in QApplicationPrivate::notify_helper(QObject*, QEvent*) + 211
#14   Object "QtWidgets", at 0x10eecb0b2, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**) + 5704
#13   Object "QtWidgets", at 0x10eecca22, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**) + 12216
#12   Object "QtCore", at 0x10f7ee40d, in QCoreApplication::notifyInternal2(QObject*, QEvent*) + 137
#11   Object "QtWidgets", at 0x10ee93191, in QApplication::notify(QObject*, QEvent*) + 2951
#10   Object "QtWidgets", at 0x10ee91995, in QApplicationPrivate::notify_helper(QObject*, QEvent*) + 211
#9    Object "plotjuggler", at 0x10e225527, in QwtPlot::event(QEvent*) + 23
#8    Object "QtWidgets", at 0x10ef2baf6, in QFrame::event(QEvent*) + 42
#7    Object "QtWidgets", at 0x10eeb9212, in QWidget::event(QEvent*) + 2246
#6    Object "plotjuggler", at 0x10e1e8a89, in PJ::PlotWidgetBase::QwtPlotPimpl::dropEvent(QDropEvent*) + 41
#5    Object "plotjuggler", at 0x10e1e8986, in std::_Function_handler<void (QEvent*), PJ::PlotWidgetBase::PlotWidgetBase(QWidget*)::{lambda(QEvent*)#1}>::_M_invoke(std::_Any_data const&, QEvent*&&) + 54
#4    Object "libstdc++.6.dylib", at 0x10faa5892, in __dynamic_cast + 130
#3    Object "libc++abi.dylib", at 0x7fff206df557, in __cxxabiv1::__si_class_type_info::has_unambiguous_public_base(__cxxabiv1::__dynamic_cast_info*, void*, int) const + 5
#2    Object "libsystem_platform.dylib", at 0x7fff20760d7c, in _sigtramp + 28
#1    Object "plotjuggler", at 0x10e184a53, in backward::SignalHandling::sig_handler(int, __siginfo*, void*) + 19
#0    Object "plotjuggler", at 0x10e184001, in backward::SignalHandling::handleSignal(int, __siginfo*, void*) [clone .isra.0] + 81
[1]    84816 segmentation fault  plotjuggler

facontidavide, 2025-12-23:
Is this still valid with the latest version?

facontidavide, 2026-09-20:
A tester confirmed dragging into the plot area works on current macOS with a recent build, so this environment-specific crash is no longer reproducible, and PlotJuggler 3 is end-of-life. PlotJuggler 4's macOS support is being reworked separately (see #1412) — please open a fresh report if you hit this there.

---
*PlotJuggler 4 is now the official release and the default branch of this repo; PlotJuggler 3 is end-of-life and no longer receives features or fixes. If this still affects you in PJ4, please reopen or file a fresh issue — it will be looked at.*

closed 2026-09-20 by facontidavide