PlotJuggler/PlotJuggler #912: Problem loading timestamps from a Header-only message

peci1, 2024-01-12:
Thanks for contributing to PlotJuggler. You are great!

# Problem description

With ros-noetic-plotjuggler-ros 2.0, I can't load any bag containing http://docs.ros.org/en/noetic/api/cras_msgs/html/msg/Heartbeat.html (install ros-noetic-cras-msgs).

The workaround is to uncheck using embedded timestamps. Or downgrading to ros-noetic-plotjuggler-ros 1.7.0 .

# Steps to reproduce (important)

- Ubuntu Focal + ROS Noetic
- Try opening [2024-01-12-13-21-23.zip](https://github.com/facontidavide/PlotJuggler/files/13919026/2024-01-12-13-21-23.zip).
- An error is shown: 

> The plugin [DataLoad ROS bags] thrown the following exception: 
> Variant::convert -> cannot convert type16

I tried to debug it, but the parsing seems complex to me. Maybe @facontidavide will have a quicker way to solve it? :)

peci1, 2024-01-12:
The error stems from here:

https://github.com/facontidavide/PlotJuggler/blob/153a9c938ab17e700066297a4a840ffd8e951933/plotjuggler_plugins/ParserROS/ros_parser.cpp#L108

facontidavide, 2024-01-12:
Give me a rosbag to reproduce the issue, please

facontidavide, 2024-01-15:
fixed. thank for reporting

closed 2024-01-15 by facontidavide

peci1, 2024-01-15:
Thanks, I've sent you a coffee's worth on Sponsors ;)

facontidavide, 2024-01-16:
To dowload the latest version RIGHT now, you can use snap:

https://snapcraft.io/plotjuggler-ros

This is the version with "embedded Noetic". The ROS debian packages usually take 2-4 weeks to be updated