PlotJuggler/PlotJuggler #914: No executable found

davide-cappelletto, 2024-01-17:
Thanks for contributing to PlotJuggler. You are great!

I cannot run plotjuggler on ros2 humble after I tried to uninstall it and reinstall it

Describe the problem and the expected behavior. Use screenshots of video if approriate.

1. tried to remove plotjuggler by executing: sudo rm -rf /opt/ros/humble/lib/plotjuggler
2. tried to reinstall it by executing: sudo snap install plotjuggler
2a. when trying to install it through "sudo apt install ros-humble-plotjuggler-ros" it gives me: ros-humble-plotjuggler-ros is already the newest version (2.0.0-3jammy.20240103.223923).
2b. But when executing "which plotjuggler", I get: /snap/bin/plotjuggler
3. after the snap install, I execute "ros2 run plotjuggler plotjuggler" and I get the message "no executable found".

- Describe your platform / Operative System.
- ROS2 WSL on Windows 10, installation from Debian packages, Ubuntu 22.04

facontidavide, 2024-01-17:
> tried to remove plotjuggler by executing: sudo rm -rf /opt/ros/humble/lib/plotjuggler

That is a brutal way to do it. `apt remove`, next time?

When you use snap, it is NOT a ROS package anymore and you DO NOT need **ros-humble-plotjuggler-ros**.

Just type `plotjuggler` in your console and...

[image]

facontidavide, 2024-01-17:
by the way, this was explained in the README.

davide-cappelletto, 2024-01-17:
Thank you, sir. I did not know that. I'm still pretty green with ubuntu, ros2 and all that.

facontidavide, 2024-01-17:
ma figurati :wink: