PlotJuggler/PlotJuggler #915: Plotjuggler misreads the information from topic /diff_drive_controller/odom and others in gazebo

PeppePercival, 2024-01-17:
Thanks for contributing to PlotJuggler. You are great!

I want to plot the data coming from a simulation in gazebo environment of a mobile robot but Plotjuggler gives me unexpected values for the topics

the information come from nav_msgs/Odometry and are the x, y positions and yaw angle. I created an extended kalman filter to localize the robot but the 3 topics /diff_drive_controller/odom, /ground_truyh and my /ekf show strange behaviour: x should be -2.0 at the beginning but for every topic in Plotjuggler il like 30 digits number. y is 0 and not changing. z seems to be y.
When i run the topic echo, the values are as expected, for example:

header:
  stamp:
    sec: 1944
    nanosec: 210000000
  frame_id: odom
child_frame_id: base_footprint
pose:
  pose:
    position:
      x: -0.8043595877305626
      y: -0.014738426840619831
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: 0.2588172129759543
      w: 0.9659263172040399

# Steps to reproduce (important)

- ROS2 in VScode
- to reproduce i don't know it that opening sim in gazebo with : <ros2 launch turtlebot3_bringup gazebo.launch.py start_rviz:=true> previously exporting the model with <export TURTLEBOT3_MODEL=burger> and see if plotting the data with plotjuggler works
- If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.

[image]
[image]

facontidavide, 2024-01-17:
which version are you using? that is fixed in 3.8.6, in theory.

Also, for future references: I think you missed this

> If it can't be reproduced with the dummy data, please share the CSV file or the rosbag that can be used to reproduce the problem.

Closing unless you confirm the problem exist in master

closed 2024-01-17 by facontidavide

facontidavide, 2024-01-17:
[image]

PeppePercival, 2024-01-17:
Oddly enough, when Installing from ''sudo apt install ros-$ROS_DISTRO-plotjuggler-ros'' it installs 3.8.5 version. Most recent version with ''sudo snap install plotjuggler''

facontidavide, 2024-01-18:
It is not odd at all. Once I submit my changes to ROS distro, it may take 2 or 3 weeks before the Open Robotics people generate a release.

This is the reason I suggest using snap instead: once I release, the new version is available within minutes