PlotJuggler/PlotJuggler #917: Segmentation fault when clicking the button to load a layout

amilcarlucas, 2024-01-18:
# Problem description
``` 
amilcar@amilcar-dev:~/iav/plotjuggler_ws/install/bin$ ./plotjuggler 
Loading compatible plugins from directory:  "/usr/local/bin/"
"libDataAPBin.so is a DataLoader plugin"
"libDataLoadCSV.so is a DataLoader plugin"
"libDataLoadULog.so is a DataLoader plugin"
"libDataStreamSample.so is a DataStreamer plugin"  ...disabled, unless option -t is used
"libDataStreamUDP.so is a DataStreamer plugin"
"libDataStreamWebSocket.so is a DataStreamer plugin"
"libDataStreamZMQ.so is a DataStreamer plugin"
"libPublisherCSV.so is a StatePublisher plugin"
Number of plugins loaded:  7 

Loading compatible plugins from directory:  "/home/amilcar/iav/plotjuggler_ws/install/bin"
"libDataLoadCSV.so is a DataLoader plugin"  ...skipping, because already loaded
"libDataLoadMCAP.so is a DataLoader plugin"
"libDataLoadULog.so is a DataLoader plugin"  ...skipping, because already loaded
"libDataStreamMQTT_Mosquitto.so is a DataStreamer plugin"
"libDataStreamSample.so is a DataStreamer plugin"  ...disabled, unless option -t is used
"libDataStreamUDP.so is a DataStreamer plugin"  ...skipping, because already loaded
"libDataStreamWebSocket.so is a DataStreamer plugin"  ...skipping, because already loaded
"libDataStreamZMQ.so is a DataStreamer plugin"  ...skipping, because already loaded
"libParserDataTamer.so is a MessageParser plugin"
"libParserROS1.so is a MessageParser plugin"
"libParserROS2.so is a MessageParser plugin"
"libProtobufParser.so is a MessageParser plugin"
"libPublisherCSV.so is a StatePublisher plugin"  ...skipping, because already loaded
QtAV 1.13.0(Apr 10 2020, 19:10:20)
Multimedia framework base on Qt and FFmpeg.
Distributed under the terms of LGPLv2.1 or later.
Shanghai, China Copyright (C) 2012-2019 Wang Bin (aka. Lucas Wang) wbsecg1@gmail.com
Donate: http://qtav.org/donate.html
Source: https://github.com/wang-bin/QtAV
Home page: http://qtav.org
"libPublisherVideoViewer.so is a StatePublisher plugin"
Type conversion already registered from type QString to type QwtText
"libToolboxFFT.so is a Toolbox plugin"
"libToolboxLuaEditor.so is a Toolbox plugin"
Type conversion already registered from type QString to type QwtText
"libToolboxQuaternion.so is a Toolbox plugin"
Number of plugins loaded:  10 

Loading compatible plugins from directory:  "/home/amilcar/.local/share/PlotJuggler"
Number of plugins loaded:  0 

Stack trace (most recent call last):
#31   Object "/home/amilcar/iav/plotjuggler_ws/install/bin/plotjuggler", at 0x5562fa5495ed, in _start
#30   Source "../csu/libc-start.c", line 308, in __libc_start_main [0x7fab45683082]
#29   Source "/home/amilcar/iav/plotjuggler_ws/src/PlotJuggler/plotjuggler_app/main.cpp", line 465, in main [0x5562fa547b94]
        462:     window->on_buttonStreamingStart_clicked();
        463:   }
        464: 
      > 465:   return app.exec();
        466: }
#28   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.12.8", at 0x7fab45e2c115, in QCoreApplication::exec()
#27   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.12.8", at 0x7fab45e243aa, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
#26   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.12.8", at 0x7fab45e7d434, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#25   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.6400.6", at 0x7fab44d954a2, in g_main_context_iteration
#24   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.6400.6", at 0x7fab44d953ff, in 
#23   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.6400.6", at 0x7fab44d9517c, in g_main_context_dispatch
#22   Object "/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5.12.8", at 0x7fab41fb032d, in 
#21   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5.12.8", at 0x7fab461ea35a, in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#20   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5.12.8", at 0x7fab4621010a, in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*)
#19   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5.12.8", at 0x7fab4620e7d2, in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*)
#18   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.12.8", at 0x7fab45e25809, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#17   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8", at 0x7fab468470ef, in QApplication::notify(QObject*, QEvent*)
#16   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8", at 0x7fab4683da65, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#15   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8", at 0x7fab4689f1eb, in 
#14   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8", at 0x7fab4689c35c, in 
#13   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8", at 0x7fab46846456, in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool)
#12   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.12.8", at 0x7fab45e25809, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#11   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8", at 0x7fab46847342, in QApplication::notify(QObject*, QEvent*)
#10   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8", at 0x7fab4683da65, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#9    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8", at 0x7fab468802b5, in QWidget::event(QEvent*)
#8    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8", at 0x7fab46934034, in QAbstractButton::mouseReleaseEvent(QMouseEvent*)
#7    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8", at 0x7fab46933e72, in 
#6    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8", at 0x7fab46932a2d, in 
#5    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8", at 0x7fab46932805, in QAbstractButton::clicked(bool)
#4    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.12.8", at 0x7fab45e5128b, in QMetaObject::activate(QObject*, int, int, void**)
#3    Source "/home/amilcar/iav/plotjuggler_ws/build/PlotJuggler/plotjuggler_app/plotjuggler_autogen/EWIEGA46WW/moc_mainwindow.cpp", line 514, in qt_metacall [0x5562fa54e049]
        511:         return _id;
        512:     if (_c == QMetaObject::InvokeMetaMethod) {
        513:         if (_id < 66)
      > 514:             qt_static_metacall(this, _c, _id, _a);
        515:         _id -= 66;
        516:     } else if (_c == QMetaObject::RegisterMethodArgumentMetaType) {
        517:         if (_id < 66)
#2    Source "/home/amilcar/iav/plotjuggler_ws/build/PlotJuggler/plotjuggler_app/plotjuggler_autogen/EWIEGA46WW/moc_mainwindow.cpp", line 422, in qt_static_metacall [0x5562fa54c2e8]
        419:         case 61: _t->on_buttonRecentLayout_clicked(); break;
        420:         case 62: _t->on_pushButtonLoadLayout_clicked(); break;
        421:         case 63: _t->on_pushButtonSaveLayout_clicked(); break;
      > 422:         case 64: _t->on_pushButtonLoadDatafile_clicked(); break;
        423:         case 65: _t->on_actionColorMap_Editor_triggered(); break;
        424:         default: ;
        425:         }
#1  | Source "/home/amilcar/iav/plotjuggler_ws/src/PlotJuggler/plotjuggler_app/mainwindow.cpp", line 2982, in QString
    |  2980:   {
    |  2981:     DataLoaderPtr loader = it.second;
    | >2982:     for (QString extension : loader->compatibleFileExtensions())
    |  2983:     {
    |  2984:       extensions.insert(extension.toLower());
      Source "/usr/include/x86_64-linux-gnu/qt5/QtCore/qstring.h", line 700, in on_pushButtonLoadDatafile_clicked [0x5562fa5a004f]
        697: #endif
        698: #if !defined(QT_NO_CAST_FROM_ASCII) && !defined(QT_RESTRICTED_CAST_FROM_ASCII)
        699:     inline QT_ASCII_CAST_WARN QString(const char *ch)
      > 700:         : d(fromAscii_helper(ch, ch ? int(strlen(ch)) : -1))
        701:     {}
        702:     inline QT_ASCII_CAST_WARN QString(const QByteArray &a)
        703:         : d(fromAscii_helper(a.constData(), qstrnlen(a.constData(), a.size())))
#0    Source "../sysdeps/x86_64/multiarch/strlen-avx2.S", line 65, in __strlen_avx2 [0x7fab457e7915]
Segmentation fault (Address not mapped to object [0x100000001])
Segmentation fault (core dumped)
```

# Answer these questions (or I will ignore your issue)

1. I use Ubuntu 20.04 and compiled the latest plotjuggler master branch from source using the instructions from the compile.md file
2. I get the crash when I click load layout.

facontidavide, 2024-01-18:
Thanks , let me check that

facontidavide, 2024-01-18:
1. Can the problem be reproduced with dummy data?
2. Send me the layout file
3 try this change, and tell me what you see. ( `MainWindow::on_pushButtonLoadDatafile_clicked()`)

```cpp
  for (auto& it : _data_loader)
  {
    DataLoaderPtr loader = it.second;
    for (const char* extension : loader->compatibleFileExtensions())
    {
      qDebug() << it.first << " ...";
      qDebug() << QString(extension);
      extensions.insert(QString(extension).toLower());
    }
  }
```

amilcarlucas, 2024-01-18:
I do not even choose the layout file. I crahes before that.
I de-installed the apt installed ros-noetic-plotjuggler version and it still crahes.

amilcarlucas, 2024-01-18:
```
amilcar@amilcar-dev:~/iav/plotjuggler_ws/install/bin$ ./plotjuggler 
Loading compatible plugins from directory:  "/usr/local/bin/"
"libDataAPBin.so is a DataLoader plugin"
"libDataLoadCSV.so is a DataLoader plugin"
"libDataLoadULog.so is a DataLoader plugin"
"libDataStreamSample.so is a DataStreamer plugin"  ...disabled, unless option -t is used
"libDataStreamUDP.so is a DataStreamer plugin"
"libDataStreamWebSocket.so is a DataStreamer plugin"
"libDataStreamZMQ.so is a DataStreamer plugin"
"libPublisherCSV.so is a StatePublisher plugin"
Number of plugins loaded:  7 

Loading compatible plugins from directory:  "/home/amilcar/iav/plotjuggler_ws/install/bin"
"libDataLoadCSV.so is a DataLoader plugin"  ...skipping, because already loaded
"libDataLoadMCAP.so is a DataLoader plugin"
"libDataLoadULog.so is a DataLoader plugin"  ...skipping, because already loaded
"libDataStreamMQTT_Mosquitto.so is a DataStreamer plugin"
"libDataStreamSample.so is a DataStreamer plugin"  ...disabled, unless option -t is used
"libDataStreamUDP.so is a DataStreamer plugin"  ...skipping, because already loaded
"libDataStreamWebSocket.so is a DataStreamer plugin"  ...skipping, because already loaded
"libDataStreamZMQ.so is a DataStreamer plugin"  ...skipping, because already loaded
"libParserDataTamer.so is a MessageParser plugin"
"libParserROS1.so is a MessageParser plugin"
"libParserROS2.so is a MessageParser plugin"
"libProtobufParser.so is a MessageParser plugin"
"libPublisherCSV.so is a StatePublisher plugin"  ...skipping, because already loaded
QtAV 1.13.0(Apr 10 2020, 19:10:20)
Multimedia framework base on Qt and FFmpeg.
Distributed under the terms of LGPLv2.1 or later.
Shanghai, China Copyright (C) 2012-2019 Wang Bin (aka. Lucas Wang) wbsecg1@gmail.com
Donate: http://qtav.org/donate.html
Source: https://github.com/wang-bin/QtAV
Home page: http://qtav.org
"libPublisherVideoViewer.so is a StatePublisher plugin"
Type conversion already registered from type QString to type QwtText
"libToolboxFFT.so is a Toolbox plugin"
"libToolboxLuaEditor.so is a Toolbox plugin"
Type conversion already registered from type QString to type QwtText
"libToolboxQuaternion.so is a Toolbox plugin"
Number of plugins loaded:  10 

Loading compatible plugins from directory:  "/home/amilcar/.local/share/PlotJuggler"
Number of plugins loaded:  0 

Stack trace (most recent call last):
#31   Object "/home/amilcar/iav/plotjuggler_ws/install/bin/plotjuggler", at 0x55b9253c261d, in _start
#30   Source "../csu/libc-start.c", line 308, in __libc_start_main [0x7f78fc934082]
#29   Source "/home/amilcar/iav/plotjuggler_ws/src/PlotJuggler/plotjuggler_app/main.cpp", line 465, in main [0x55b9253c0bc4]
        462:     window->on_buttonStreamingStart_clicked();
        463:   }
        464: 
      > 465:   return app.exec();
        466: }
#28   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.12.8", at 0x7f78fd0dd115, in QCoreApplication::exec()
#27   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.12.8", at 0x7f78fd0d53aa, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
#26   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.12.8", at 0x7f78fd12e434, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#25   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.6400.6", at 0x7f78fc0464a2, in g_main_context_iteration
#24   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.6400.6", at 0x7f78fc0463ff, in 
#23   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.6400.6", at 0x7f78fc04617c, in g_main_context_dispatch
#22   Object "/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5.12.8", at 0x7f78f926132d, in 
#21   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5.12.8", at 0x7f78fd49b35a, in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#20   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5.12.8", at 0x7f78fd4c110a, in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*)
#19   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5.12.8", at 0x7f78fd4bf7d2, in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*)
#18   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.12.8", at 0x7f78fd0d6809, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#17   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8", at 0x7f78fdaf80ef, in QApplication::notify(QObject*, QEvent*)
#16   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8", at 0x7f78fdaeea65, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#15   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8", at 0x7f78fdb501eb, in 
#14   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8", at 0x7f78fdb4d35c, in 
#13   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.12.8", at 0x7f78fdaf7456, in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool)
[... 18 lines cut ...]
        419:         case 61: _t->on_buttonRecentLayout_clicked(); break;
        420:         case 62: _t->on_pushButtonLoadLayout_clicked(); break;
        421:         case 63: _t->on_pushButtonSaveLayout_clicked(); break;
      > 422:         case 64: _t->on_pushButtonLoadDatafile_clicked(); break;
        423:         case 65: _t->on_actionColorMap_Editor_triggered(); break;
        424:         default: ;
        425:         }
[... cut ...]
        699:     inline QT_ASCII_CAST_WARN QString(const char *ch)
      > 700:         : d(fromAscii_helper(ch, ch ? int(strlen(ch)) : -1))
        701:     {}
        702:     inline QT_ASCII_CAST_WARN QString(const QByteArray &a)
        703:         : d(fromAscii_helper(a.constData(), qstrnlen(a.constData(), a.size())))
#0    Source "../sysdeps/x86_64/multiarch/strlen-avx2.S", line 96, in __strlen_avx2 [0x7f78fca98947]
Segmentation fault (Signal sent by the kernel [(nil)])
Segmentation fault (core dumped)
```

amilcarlucas, 2024-01-18:
I've found a plotjuggler 3.2.0 on /usr/local/bin/plotjuggler 
Could that be the source of the problems?

facontidavide, 2024-01-18:
I think I see the problem. If you check the log at the beginning, you are loading the plugins inside **/usr/local/bin/**, and skipping he one inside **/home/amilcar/iav/plotjuggler_ws/install/bin**.

This is likely the issue

amilcarlucas, 2024-01-23:
OK, I deleted all versions except one, and now it works. Closing

closed 2024-01-23 by amilcarlucas