PlotJuggler/PlotJuggler #918: plotjuggler 3.5.0/3.6.0 startup issue on WindowsOS

m-yoshida04, 2024-01-19:
Hi, I installed plotjuggler 3.6.0 from [here](https://github.com/facontidavide/PlotJuggler/releases/download/3.6.0/PlotJuggler-Windows-3.6.0-installer.exe). Software installation has been successfully completed, however plotjuggler does not startup (i.e. nothing happens) even after clicking the "plotjuggler.exe" in order to launch the plotjuggler.
Then, I uninstalled 3.6.0 and installed 3.5.0 from [here](https://github.com/facontidavide/PlotJuggler/releases/download/3.5.0/PlotJuggler-Windows-3.5.0-installer.exe). 
But it did not work as well...

- Operating System:    Windows 10
- PlotJuggler Verison:  3.5.0 and 3.6.0

it would be great if you let me know how this issue can be solved.
Thank you!

facontidavide, 2024-01-19:
I need to create a new installer

m-yoshida04, 2024-01-19:
Thank you for reply.
Do you mean the current installer only works for Windows 11 and does not work for Windows 10?

facontidavide, 2024-02-22:
Tru latest Windows installer

closed 2024-02-22 by facontidavide