PlotJuggler/PlotJuggler #924: Segmentation fault when loading bagfiles

siferati, 2024-01-24:
# Problem description

Segmentation fault when loading bagfiles:

```
Stack trace (most recent call last):
#31   Object "/snap/plotjuggler-ros/11/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.6400.6", at 0x7fd2c73923ff, in 
#30   Object "/snap/plotjuggler-ros/11/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.6400.6", at 0x7fd2c739217c, in g_main_context_dispatch
#29   Object "/snap/plotjuggler-ros/11/kf5/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5.15.7", at 0x7fd2c4609e5d, in 
#28   Object "/snap/plotjuggler-ros/11/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5.15.7", at 0x7fd2c8b176db, in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#27   Object "/snap/plotjuggler-ros/11/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5.15.7", at 0x7fd2c8b40e84, in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*)
#26   Object "/snap/plotjuggler-ros/11/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5.15.7", at 0x7fd2c8b3f7df, in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*)
#25   Object "/snap/plotjuggler-ros/11/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.7", at 0x7fd2c871a0a9, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#24   Object "/snap/plotjuggler-ros/11/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.7", at 0x7fd2c92adc67, in QApplication::notify(QObject*, QEvent*)
#23   Object "/snap/plotjuggler-ros/11/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.7", at 0x7fd2c92a4e52, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#22   Object "/snap/plotjuggler-ros/11/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.7", at 0x7fd2c93062d3, in 
#21   Object "/snap/plotjuggler-ros/11/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.7", at 0x7fd2c9302fb3, in 
#20   Object "/snap/plotjuggler-ros/11/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.7", at 0x7fd2c92ad156, in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool)
#19   Object "/snap/plotjuggler-ros/11/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.7", at 0x7fd2c871a0a9, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#18   Object "/snap/plotjuggler-ros/11/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.7", at 0x7fd2c92ade56, in QApplication::notify(QObject*, QEvent*)
#17   Object "/snap/plotjuggler-ros/11/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.7", at 0x7fd2c92a4e52, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#16   Object "/snap/plotjuggler-ros/11/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.7", at 0x7fd2c9445dc2, in QPushButton::event(QEvent*)
#15   Object "/snap/plotjuggler-ros/11/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.7", at 0x7fd2c92e7dc5, in QWidget::event(QEvent*)
#14   Object "/snap/plotjuggler-ros/11/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.7", at 0x7fd2c939ef24, in QAbstractButton::mouseReleaseEvent(QMouseEvent*)
#13   Object "/snap/plotjuggler-ros/11/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.7", at 0x7fd2c939ed62, in 
#12   Object "/snap/plotjuggler-ros/11/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.7", at 0x7fd2c939d88d, in 
#11   Object "/snap/plotjuggler-ros/11/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.7", at 0x7fd2c939d615, in QAbstractButton::clicked(bool)
#10   Object "/snap/plotjuggler-ros/11/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.7", at 0x7fd2c875137b, in 
#9    Object "/snap/plotjuggler-ros/11/usr/local/bin/plotjuggler", at 0x56397a679749, in MainWindow::qt_metacall(QMetaObject::Call, int, void**)
#8    Object "/snap/plotjuggler-ros/11/usr/local/bin/plotjuggler", at 0x56397a6774b8, in MainWindow::qt_static_metacall(QObject*, QMetaObject::Call, int, void**)
#7    Source "/usr/include/c++/9/bits/stl_vector.h", line 1756, in MainWindow::on_pushButtonLoadDatafile_clicked() [0x56397a6d2455]
#6    Source "/usr/include/c++/9/bits/stl_vector.h", line 1756, in MainWindow::loadDataFromFiles(QStringList) [0x56397a6cd435]
#5    Source "/usr/include/c++/9/bits/stl_vector.h", line 1756, in MainWindow::loadDataFromFile[abi:cxx11](PJ::FileLoadInfo const&) [0x56397a6c308e]
#4    Source "/root/parts/plotjuggler-ros/src/plotjuggler-ros-plugins/src/DataLoadROS/dataload_ros.cpp", line 123, in readDataFromFile [0x7fd2c01ab270]
#3  | Source "/root/parts/plotjuggler-ros/src/plotjuggler-ros-plugins/src/DataLoadROS/dataload_ros.cpp", line 64, in CreateParserROS
      Source "/root/parts/plotjuggler-ros/src/plotjuggler-ros-plugins/src/ros_parsers/ros1_parser.h", line 13, in getAllTopics [0x7fd2c01aaeb8]
#2    Object "/snap/plotjuggler-ros/11/usr/local/bin/libParserROS1.so", at 0x7fd2a1d4e736, in ParserFactoryROS1::createParser(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, PJ::PlotDataMapRef&)
#1    Object "/snap/plotjuggler-ros/11/usr/local/bin/libParserROS1.so", at 0x7fd2a1d506aa, in ParserROS::ParserROS(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, RosMsgParser::Deserializer*, PJ::PlotDataMapRef&)
#0    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28", at 0x7fd2c837d6c4, in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::compare(char const*) const
Segmentation fault (Address not mapped to object [0x30])
/snap/plotjuggler-ros/11/usr/bin/launcher-plotjuggler-ros: line 15: 49536 Segmentation fault      (core dumped) ${SNAP}/usr/local/bin/plotjuggler --plugin_folders $PLUGIN_FOLDERS $@
```

# Answer these questions (or I will ignore your issue)

> Describe your platform / Operative System and the **version of PlotJuggler** you are using.

PlotJuggler-3 3.8.8 on ubuntu 20 with ros noetic.
PlotJuggler was installed via snap

> Compile and try the latest version in the **main** branch and check if the problem persists.

Yes still have segmentation fault even when compiling manually

> Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".

No, the issue doesn't happen with bagfiles in samples folder

> If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.

Seems like the issue is related with the number of topics in the bagfile? Or maybe amount of data? If I keep removing topics from this bagfile eventually it loads correctly.

[bagfile.zip](https://github.com/facontidavide/PlotJuggler/files/14031705/bagfile.zip)

facontidavide, 2024-01-24:
Thanks.I will investigate

facontidavide, 2024-01-24:
Problem found. It crashes when it finds a message with type `std_msgs/Empty`. Pushing a solution

siferati, 2024-01-25:
Thanks, I compiled locally and it solved the issue!

However, I have a different issue now that seems related -- only happens with `std_msgs/Empty` topic:

[image]

I'm uploading 2 bagfiles below. Both of them contain only `std_msgs/Empty` topic, but one of them loads correctly and the other one gives the above error.

- [bad.zip](https://github.com/facontidavide/PlotJuggler/files/14045233/bad.zip)
- [good.zip](https://github.com/facontidavide/PlotJuggler/files/14045234/good.zip)

facontidavide, 2024-01-25:
The empty message is definitely a corner case I did not consider. Also... Being empty, you will not be able to see it in plotjuggler

commit d6f18b05 (2024-01-26): support std_msgs/Empty (fix #924)
closed 2024-01-26 by facontidavide