PlotJuggler/PlotJuggler #926: Bugs in csv plots - wrong axis labels and wrong plot on non-decimal x axis numbers

ipa-jsk, 2024-01-26:
# Problem description

Thank you for the amazing plotjuggler tool. I've been using it for several years now and it was very helpful all the time.
But, in the current version of plotjuggler, I cannot plot the attached [test.csv](https://github.com/facontidavide/PlotJuggler/files/14062954/test.csv) file (and similar files). I tried the samples, and could reproduce it by modifying the simple.csv file from datasamples folder. I chose the timestamp row to be the x axis on import.

In [dummy.csv](https://github.com/facontidavide/PlotJuggler/files/14062902/dummy.csv), I changed the first row to start from `10.0` instead of `0.0`. result:

[image]

The x axis labels are not correct.

Then, I changed the first row to start from 1 to 400 and removed the decimal point. (
[dummy_fail.csv](https://github.com/facontidavide/PlotJuggler/files/14062938/dummy_fail.csv)) Result:

[image]

Something weird happens, which is the same like in my test.csv file where I started..

# Answer these questions (or I will ignore your issue)

1. Describe your platform / Operative System and the **version of PlotJuggler** you are using.
     Ubuntu 20.04.06 LTS: 
[image]

    Plotjuggler Version 3.8.9 installed via snap. Maybe its worth saying that I'm from Germany, if this could be some decimal point locale related issue...
3. Compile and try the latest version in the **main** branch and check if the problem persists.
   Since there is only one more commit on the main branch (3be95bbb0318f7944456870cf148b965532abef1), which is only related to ros messages, I didn't do that.
5. Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".
    -> see above.
7. If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.
    -> see above.

facontidavide, 2024-01-26:
Yes, I am aware. I will try to fix it today

facontidavide, 2024-01-26:
About the dummy.csv, I think you missed this:

[image]

facontidavide, 2024-01-26:
issue fixed and solution pushed

closed 2024-01-26 by facontidavide