PlotJuggler/PlotJuggler #928: Cannot edit properties of XY plot if multiple sources are loaded with prefix

peci1, 2024-01-26:
# Problem description

Describe the problem and the expected behavior. Use screenshots or a video, if applicable.

# Answer these questions (or I will ignore your issue)

1. Describe your platform / Operative System and the **version of PlotJuggler** you are using.
2. Compile and try the latest version in the **main** branch and check if the problem persists.
3. Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".
4. If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.

When I open two bagfiles at once (with prefix) and create an XY plot from the loaded data, right-clicking and Edit curves shows this warning:

facontidavide, 2024-01-29:
the explanation is pretty simple: you can not mix two timeseries in a XY plot unless they have EXACTLY the same timestamps and number of elements.

Not your case

closed 2024-01-29 by facontidavide

peci1, 2024-01-29:
I don't think you got it right. It suffices to add one xy series to the plot to trigger this issue. That still seems to be a bug to me.