PlotJuggler/PlotJuggler #929: Floating point truncated exception

siferati, 2024-01-29:
# Problem description

When selecting topics to replay I get a floating point truncated exception

[image]

# Answer these questions (or I will ignore your issue)

> Describe your platform / Operative System and the **version of PlotJuggler** you are using.

PlotJuggler-3 3.8.10 on ubuntu 20 with ros noetic.
Compiled from source.

> Compile and try the latest version in the **main** branch and check if the problem persists.

Yes.

> Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".

Doesn't happen with those bag files.

> If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.

Minimum reproducible example with a single message.
When you open it in plot juggler and select this topic you should see the above error.
[float.zip](https://github.com/facontidavide/PlotJuggler/files/14078332/float.zip)

facontidavide, 2024-01-29:
This is a well know issue (duplicated many many times...). 

I will just remove this and then people will complain about their long integers being not the expected one

closed 2024-01-29 by facontidavide

facontidavide, 2024-01-29:
I am quite tired of answering the same question over and over again, andthat people don't search for similar questions before asking.
Therefore I "fixed" the problem.
[image]

Enjoy your wrong number 

[image]