PlotJuggler/PlotJuggler #930: plotjuggler topic republisher issue: segmentation fault

InstaJun, 2024-02-01:
# Problem description

 I updated the plotjuggler from version 3.6.0 to version 3.8.8 on my native linux ubunbu 20.04 machine. I also updated the current version of plotjuggler-ros-plugins. When I tried to load rosbag file like before, I got a segmentation fault on the terminal. The plotjugger was then dead. 
I solved this problem by changing the current version of plotjuggler-ros-plugins to the old one. 

Anyone has the same problem like me?

facontidavide, 2024-02-01:
I will be happy to fix this ASAP. Can you share the rosbag that causes the issue?
Also, what happen is you use the latest version of PlotJuggler on master branch?
Way be that wass fixed already

facontidavide, 2024-02-01:
I don't understand: is the problem related to loading the rosbag or republishing?

InstaJun, 2024-02-01:
> I will be happy to fix this ASAP. Can you share the rosbag that causes the issue? Also, what happen is you use the latest version of PlotJuggler on master branch? Way be that wass fixed already

I will check the latest version on master branch after work. Hope it is fixed already.

InstaJun, 2024-02-01:
> I don't understand: is the problem related to loading the rosbag or republishing?
Sorry for the misunderstanding, it is related to loading the rosbag. I could not load the rosbag after the rosbag file is checked. The window of loading process went black and the terminal showed me a segmentation fault.

facontidavide, 2024-02-01:
>  I could not load the rosbag after the rosbag file is checked. 

What do you mean by "checked"? I don't understand. Please share the rosbag causing the issue

InstaJun, 2024-02-02:
I have tried the latest version plotjuggler on the main branch, it seems the problem I had was fixed.

closed 2024-02-02 by InstaJun