PlotJuggler/PlotJuggler #935: Crash after opening layout file and starting UDP Server Streamer

mmichini, 2024-02-14:
Hi - thanks for the amazing tool!

# Problem description

Plotjugger crashes when opening the attached layout file and enabling json UDP streaming with "Start the previously used streaming plugin? UDP Server". Tested on `main` commit `8a297ad689d385524f9ef221aec82b177691fdf9`.

Repro steps:

1. Open plotjuggler
2. Load attached layout file
3. When prompted "Start the previously used streaming plugin? UDP Server", click "Yes"
4. Use settings Address: 127.0.0.1, Port 9999, Message Protocol json, no protocol options.

Screencast video attached (let me know if this video format is an issue - it's the default Ubuntu screen capture format)

[crash_main_8a297ad689d.webm](https://github.com/facontidavide/PlotJuggler/assets/24902355/1d6d79f9-01b5-404c-a90a-fd955ec5cb80)

# Answer these questions (or I will ignore your issue)

1. Describe your platform / Operative System and the **version of PlotJuggler** you are using.

- Ubuntu 22.04.3 LTS (kernel version 6.5.0-15.15~22.04.1)
- Plotjuggler `main` `8a297ad689d385524f9ef221aec82b177691fdf9` built from source

1. Compile and try the latest version in the **main** branch and check if the problem persists.

Problem persists on latest version of `main`.

1. Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".

I'm not sure about this one, but the problem does seem to be reproduce-able with the attached layout file.

1. If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.

Layout file attached.

[crash.xml.txt](https://github.com/facontidavide/PlotJuggler/files/14284944/crash.xml.txt)

Console output from crash attached

[console_output.txt](https://github.com/facontidavide/PlotJuggler/files/14284970/console_output.txt)

mmichini, 2024-02-14:
Could this be related to https://github.com/facontidavide/PlotJuggler/issues/934?

facontidavide, 2024-02-14:
yes, it is exactly the same problem

closed 2024-02-14 by facontidavide

mmichini, 2024-02-15:
@facontidavide ah, sorry for the dupe. I thought perhaps this was a different issue related to streaming, since if I click "No" at the "Start the previously used streaming plugin? UDP Server" dialog, the crash does not occur. Instead, just a "Can't find one or more curves" dialog appears, but no crash.

facontidavide, 2024-02-15:
I pushed a potential fix... Can you try it?

mmichini, 2024-02-15:
Hi @facontidavide - I built and tested this on `2376fdf75b4e01da5d964eb9ed2260f3ad025dca` and I can no longer reproduce this issue. Thanks for the quick fix! Will let you know if I see any further issues.

commit 2376fdf7 (2024-02-14): fix #935 and #934