PlotJuggler/PlotJuggler #936: Performance loss on 3.9.0

parakhm95, 2024-02-14:
# Problem description

Starting with the recent release, not sure if it started at 3.8.x or 3.9.0, I have noticed a significant drop in performance of the plots for live plotting in ROS. The topics at 250 Hz or even lower used to be easily plotted with no hiccups but I have noticed that the plot now skips points constantly and the density of the plotted points changes if I slow down my Gazebo simulation by 2-3x. This seems to happen even before the CPU is above 70% and is a very recent phenomenon. 

I have not seen Plotjuggler miss a beat in the last 6 years even when my system is throttling and my Gazebo sim is struggling. 

# Answer these questions (or I will ignore your issue)

1. Describe your platform / Operative System and the **version of PlotJuggler** you are using.
2. Compile and try the latest version in the **main** branch and check if the problem persists.
3. Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".
4. If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.

facontidavide, 2024-02-16:
Which part of "give me a way to reproduce the issue" was unclear?

parakhm95, 2024-02-16:
I will try to design a workload that I can share and demonstrates my point but the cockiness/arrogance was uncalled for.

facontidavide, 2024-02-16:
I have tried quickly with RQT publisher, but everything looks in order

 [image]

parakhm95, 2024-02-16:
@facontidavide I will break down my response into two for addressing the bug/issue and this little run-in. 

First of all, the community is well aware of and thankful for your contributions and  plotjuggler plays an integral part in the lives of robotics community in general. The tool has been invaluable for me in the last 5-6 years in my work and I cannot overstate its importance. But, we are also aware that you have a full-time job and if this is stressful, I would suggest that a call for developers to assist you is a good way to share this burden with the open-source community.  

The reason is I ignored those instructions is because my issue is live plotting and that is not covered by the scope of the template presented. The reason was to also publicly start a discourse about this. And now, more details about the problem and my plan to recreate it and check it. 

I have some simulation environments that are very taxing and my realtime simulation rate for them usually drops below 1.0 on my work machine. I have been developing in those two particular environments for 2 years and even when the entire system was bogged down in the past, I have easily been able to live plot my ROS topics at 100Hz easily. Last week I noticed that a data stream at 100Hz and another at 250Hz were missing dots on the plot (vs time) and only when I slowed down the simulation deliberately to prevent overloading, I got higher density of points being plotted. This was the first time I have seen this on Plotjuggler. Trusting PlotJuggler as much as I do, I chased down a time-dilation issue on my side which was never to be found because it turned out that the rate and time were fine. My next suspicion was my graphics driver so I checked that as well. But I noticed that my co-worker was also confused by something similar two evenings ago where the plot density of points from a 100 Hz topic was all over the place. I revisited my old tried and tested simulation environment and I noticed hiccups that I haven't seen before. This is difficult to replicate for debugging because I would have to provide access to a demanding environment which is proprietary/closed-source, however, I will consider it. 

My plan is as follows:
* Figure out which environment can be easily obtained in the open-source community, first guess would be PX4's SITL simulation where I first noticed this. I will then consider adding you to my repository I have been developing. 
* Once established, I will progressively downgrade the plotjuggler package either through snap or by checking out tags on the master and building from source. 
* And then I will write back with my findings. 

What are your thoughts on this?

facontidavide, 2024-02-16:
Anyway, we are talking about ROS topics here, right? can it be reproduced with **rosbag play** ?

parakhm95, 2024-02-16:
I fully understand. I will let you know as I have more information during my testing. 

I will also try out the hotspot tool yes. About the rosbag play, I can try using my rosbags but I suspect that this is related to the cpu load so I would have to find a way to maybe adjust that stress while playing. Let's see.

parakhm95, 2024-02-19:
@facontidavide I already did it, sorry I broke the markdown formatting, read my comment again.  Sent you the rosbag as well. What should be the next direction of investigation? Apparently this is not CPU load dependent at all.

facontidavide, 2024-02-19:
perfect. I will come back to you with a solution!

[image]

parakhm95, 2024-02-19:
Thank you so much and please let us know what you find, I am curious (as an Engineer and a not-by-choice software developer).

facontidavide, 2024-02-19:
Solved. The problem was in https://github.com/PlotJuggler/plotjuggler-ros-plugins

It was in fact CPU bounded: a relatively expensive operation was done over and over again, when it shouldn't.

See commit here https://github.com/PlotJuggler/plotjuggler-ros-plugins/commit/35b9de91cecb4f3c75706b53f199773fc30b209d

Please try and let me know if you see the issue fixed also on your side

facontidavide, 2024-02-19:
Just curious... @matemat13 who are you and why do you feel compelled to add emoji to all the comments? :smile:

closed 2024-02-19 by facontidavide

matemat13, 2024-02-19:
> Just curious... @matemat13 who are you and why do you feel compelled to add emoji to all the comments? 😄 Nothing against it, of course!

I'm Parakh's colleague. We both run into this issue, so I wanted to support it by adding some interaction to show that it's not just Parakh who cares about it without writing redundant "me too" comments. And I didn't like your original reaction, so I felt compelled to show it as well.

matemat13, 2024-02-19:
But thank you for resolving this issue so fast, we all appreciate your hard work on this tool.