PlotJuggler/PlotJuggler #939: ULog float information printing broken

declancm, 2024-02-24:
# Problem description

ULog information messages of type float don't print correctly in the Info tab. They all display the same huge value. I have to use Pyulog to view the float information messages

[image]

# Answer these questions (or I will ignore your issue)

1. Describe your platform / Operative System and the **version of PlotJuggler** you are using.
    Windows 11
3. Compile and try the latest version in the **main** branch and check if the problem persists.
    Tested on 3.9.0 and read the release notes after - haven't been able to build master successfully.
5. Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".
    There are no ULog files in the datasamples.
7. If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.
    Can't attach ULog files.

facontidavide, 2024-02-24:
Please provide a sample file to reproduce the issue

declancm, 2024-02-25:
https://drive.google.com/file/d/1xVuKQ5Ncl0fE78Wm5t_-E2K050dptbgk/view?usp=drive_link

declancm, 2024-03-06:
I'll work on a fix

declancm, 2024-03-06:
I've found the issue - the info values for most info message types are getting cast from the key strings instead of the value. I'll submit a pull request ASAP. This is a screenshot from the working info message:

[image]

facontidavide, 2024-03-06:
oh, thanks for finding this!

closed 2024-03-08 by declancm