PlotJuggler/PlotJuggler #943: ROS Feature: plot transformations `/tf` between any couple of frames

torydebra, 2024-03-01:
Hi Davide,

It would be really nice to have the possibility to plot the `/tf` among any couple frames.
Indeed, in the tf topic the transformations are published from the father to the son link , and some computations would be necessary to have the transformations between any couple other frames.

Do you see the possiblity to have such a feature? Or a workaround to plot this?

Thanks!

facontidavide, 2024-03-01:
not impossible, but it is not on the roadmap. Maybe 2-3 days of work (that I can not do, right now)

If any company wants to sponsor this feature, let me know.

lucasw, 2024-03-09:
The work-around I've used for this (and a lot of other things) is a utility node I made called `old_tf_to_new_tf` in the `tf_demo` package where you can run a node that will look up the relationship between any two frames and publish that relationship as a new single transform on /tf (be sure to name the child differently to avoid multiple parents).  There's a python and C++ version that are mostly the same, the C++ version is faster but if you want to make modifications start with the python.

Here's an example of few getting launched:

https://github.com/lucasw/tf_demo/blob/old_tf_to_new_tf/launch/old_tf_to_new_tf.launch

facontidavide, 2026-11-09:
temporarily closing this, I am not going to do it unless it is a sponsored feature

closed 2026-11-09 by facontidavide