PlotJuggler/PlotJuggler #947: PlotJuggler 3.9.1 compile error on macOS 14.3.1

KenzoKenzo94, 2024-03-06:
I am trying to compile PlotJuggler 3.9.1 from source on my Intel Mac on macOS 14.3.1 following the [COMPILE.md](https://github.com/facontidavide/PlotJuggler/blob/main/COMPILE.md), but it looks like `brew link` is missing the `--override` flag.

```
brew link qt@5 --override
Usage: brew link, ln [options] installed_formula [...]

Symlink all of formula's installed files into Homebrew's prefix. This is done
automatically when you install formulae but can be useful for manual
installations.

      --overwrite                  Delete files that already exist in the prefix
                                   while linking.
  -n, --dry-run                    List files which would be linked or deleted
                                   by brew link --overwrite without actually
                                   linking or deleting any files.
  -f, --force                      Allow keg-only formulae to be linked.
      --HEAD                       Link the HEAD version of the formula if it is
                                   installed.
  -d, --debug                      Display any debugging information.
  -q, --quiet                      Make some output more quiet.
  -v, --verbose                    Make some output more verbose.
  -h, --help                       Show this message.
Error: invalid option: --override
```
So, I just ran it as `brew link qt@5`. But then, when I try to compile it, I get this error:

```
cmake -S src/PlotJuggler -B build/PlotJuggler -DCMAKE_INSTALL_PREFIX=install

-- The C compiler identification is AppleClang 15.0.0.15000309
-- The CXX compiler identification is AppleClang 15.0.0.15000309
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - broken
CMake Error at /usr/local/Cellar/cmake/3.28.3/share/cmake/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "/Library/Developer/CommandLineTools/usr/bin/cc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: '/Users/vico/plotjuggler_ws/build/PlotJuggler/CMakeFiles/CMakeScratch/TryCompile-duu3E6'

    Run Build Command(s): /usr/local/Cellar/cmake/3.28.3/bin/cmake -E env VERBOSE=1 /usr/bin/make -f Makefile cmTC_76107/fast
    /Library/Developer/CommandLineTools/usr/bin/make  -f CMakeFiles/cmTC_76107.dir/build.make CMakeFiles/cmTC_76107.dir/build
    Building C object CMakeFiles/cmTC_76107.dir/testCCompiler.c.o
    /Library/Developer/CommandLineTools/usr/bin/cc   -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX14.4.sdk -mmacosx-version-min=14.3 -MD -MT CMakeFiles/cmTC_76107.dir/testCCompiler.c.o -MF CMakeFiles/cmTC_76107.dir/testCCompiler.c.o.d -o CMakeFiles/cmTC_76107.dir/testCCompiler.c.o -c /Users/vico/plotjuggler_ws/build/PlotJuggler/CMakeFiles/CMakeScratch/TryCompile-duu3E6/testCCompiler.c
    Linking C executable cmTC_76107
    /usr/local/Cellar/cmake/3.28.3/bin/cmake -E cmake_link_script CMakeFiles/cmTC_76107.dir/link.txt --verbose=1
    /Library/Developer/CommandLineTools/usr/bin/cc  -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX14.4.sdk -mmacosx-version-min=14.3 -Wl,-search_paths_first -Wl,-headerpad_max_install_names /usr/local/opt/qt@5/lib  CMakeFiles/cmTC_76107.dir/testCCompiler.c.o -o cmTC_76107
    ld: file cannot be mmap()ed, errno=22 path=/usr/local/opt/qt@5/lib in '/usr/local/opt/qt@5/lib'
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    make[1]: *** [cmTC_76107] Error 1
    make: *** [cmTC_76107/fast] Error 2

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:3 (PROJECT)

-- Configuring incomplete, errors occurred!
```

I tried already reinstalling the command line tools with `xcode-select --install`, but no luck. Any idea on what could have gone wrong here?

Thanks!

ubaldot, 2024-03-07:
I had the same problem. Note that the error is the following:

`ld: file cannot be mmap()ed, errno=22 path=/opt/homebrew/opt/qt@5/lib in '/opt/homebrew/opt/qt@5/lib'`

and that happens around the 75% of the build. 

However, I managed at some point to reach 84% but then it failed again due to an include not found. 
I reached there simply by not setting the LDFLAGS variable. 
The error at 84% is about a missing include file:

`/Users/ubaldot/plotjuggler_ws/src/PlotJuggler/plotjuggler_plugins/DataStreamZMQ/zmq.hpp:36:10: fatal error: 'zmq.h' file not found`

Unfortunately I am not a cmake guy, nor have any idea on where such a `zmq.h` could be located.... Any help is appreciated.

ubaldot, 2024-03-07:
**SOLVED**
1. Add the following to your `CMakeList.txt `file:

```
# Include PkgConfig module
find_package(PkgConfig REQUIRED)

# Use pkg_check_modules to find ZeroMQ
pkg_check_modules(ZEROMQ REQUIRED IMPORTED_TARGET libzmq)

# Include ZeroMQ directories
include_directories(${ZEROMQ_INCLUDE_DIRS})

```
2. Don't export the LDFLAG environment variable i.e. use the following:

```
QT_HOME=$(brew --prefix qt@5)
export CPPFLAGS="-I $QT_HOME/include"
export PKG_CONFIG_PATH="$QT_HOME/lib/pkgconfig"
```

instead of the following (as written in the guide):

```
QT_HOME=$(brew --prefix qt@5)
export CPPFLAGS="-I $QT_HOME/include"
export PKG_CONFIG_PATH="$QT_HOME/lib/pkgconfig"
export LDFLAGS="$QT_HOME/lib"
```

facontidavide, 2024-03-08:
can you please create a PR, with the required corrections?

closed 2024-03-08 by facontidavide

ubaldot, 2024-03-08:
Sure! That I am able to do :D

lgkimjy, 2024-03-28:
@ubaldot Could you tell me which CMakeLists.txt should I modify?

ubaldot, 2024-03-28:
@lgkimjy it is written in this PR: https://github.com/facontidavide/PlotJuggler/pull/950
: