PlotJuggler/PlotJuggler #955: It only recognizes the messages of ROS2

serkanMzlm, 2024-03-18:
# Problem description
I created my own message type in ROS2 and executed `. install/setup.bash` in the terminal. I can print the published message to the screen using `ros2 topic echo`. However, when I tried to include it in the plotjuggler application in the same terminal, the topic part gave me a warning

Describe the problem and the expected behavior. Use screenshots or a video, if applicable.
[image]

# Answer these questions (or I will ignore your issue)

1. Describe your platform / Operative System and the **version of PlotJuggler** you are using.
- OS: Ubuntu 22.04 LTS (Jammy Jellyfish)
- PlotJuggler Version: 3.9.1
- ROS2: Humble
2. Compile and try the latest version in the **main** branch and check if the problem persists.
- Sure, the error persists
3. Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".
- The error is due to the topic's data type not being directly from ROS2's original data types.
4. If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.
[gps_topic.zip](https://github.com/facontidavide/PlotJuggler/files/14634727/gps_topic.zip)

serkanMzlm, 2024-03-18:
I installed  `sudo apt install ros-humble-plotjuggler-ros`, and now when I run `ros2 run plotjuggler plotjuggler` it works. However, when I run the regular `plotjuggler`, it still gives the same warning.

ndahn, 2024-04-09:
That one threw me for a loop as well... thanks!

facontidavide, 2026-01-06:
correct, this is a limitation of the snap