PlotJuggler/PlotJuggler #958: Having trouble opening an mcap file created on Linux in PlotJuggler for Windows?

Pratheeba91, 2024-04-01:
# Problem description

The .mcap files were generated using data tamer C++ libraries in Linux environment. They are opening in PlotJuggler in Linux. But the same .mcap files are not opening in the Windows version of the PlotJuggler or Fox glove studio. 
Getting this error: " The plugin [DataLoad MCAP] thrown the following exception: stoul argument out of range
Additional information:

1. Windows / Plotjuggler version 3.9.0
2. installed and tried with Windows / Plotjuggler version 3.9.0 exe .
[image]

jlack1987, 2024-07-10:
I have the same problem. Created mcap on Linux, tried to open on Windows

facontidavide, 2026-10-01:
The Windows failure is caused by reading a 64-bit schema hash with `std::stoul`: Windows uses a 32-bit `unsigned long`. The upstream fix is https://github.com/PickNikRobotics/data_tamer/pull/92. PJ4 integration is prepared in https://github.com/PlotJuggler/pj-official-plugins/pull/322, covering both the standalone DataTamer parser and ROS schema/snapshot messages. The integration is a draft until the upstream change is released and its dependency pin can use that release. Keeping this issue open until the fix ships.

facontidavide, 2026-11-04:
Sorry for the delay in following up. Development will now focus exclusively on PlotJuggler 4 (PJ4).

The Windows 64-bit schema-hash overflow is fixed in the PJ4 DataTamer update merged through [official-plugins #330](https://github.com/PlotJuggler/pj-official-plugins/pull/330), with full-width-hash regressions for both parser paths. The updates are published as [DataTamer parser 0.10.0](https://github.com/PlotJuggler/pj-official-plugins/releases/tag/parser_data_tamer/v0.10.0) and [ROS parser 1.3.1](https://github.com/PlotJuggler/pj-official-plugins/releases/tag/parser_ros/v1.3.1). Please update the relevant extension in PJ4. Old PJ3 parser binaries are unaffected by those extension releases.

closed 2026-11-04 by facontidavide