PlotJuggler/PlotJuggler #961: Compile problem with macOS 14.4.1: error: only virtual member functions can be marked 'override'                   const std::string& message) override;

RENyunfan, 2024-04-08:
# Problem description

Following #947, I still cannot compile plotjuggler on Mac with macOS 14.4.1  and Apple M3 chip. Need help.

[image]

```bash
[ 92%] Built target ProtobufParser_autogen
[ 93%] Building CXX object plotjuggler_plugins/ParserProtobuf/CMakeFiles/ProtobufParser.dir/ProtobufParser_autogen/mocs_compilation.cpp.o
In file included from /Users/renyunfan/Documents/plotjuggler_ws/build/PlotJuggler/plotjuggler_plugins/ParserProtobuf/ProtobufParser_autogen/mocs_compilation.cpp:2:
In file included from /Users/renyunfan/Documents/plotjuggler_ws/build/PlotJuggler/plotjuggler_plugins/ParserProtobuf/ProtobufParser_autogen/EWIEGA46WW/moc_protobuf_factory.cpp:10:
In file included from /Users/renyunfan/Documents/plotjuggler_ws/build/PlotJuggler/plotjuggler_plugins/ParserProtobuf/ProtobufParser_autogen/EWIEGA46WW/../../../../../../src/PlotJuggler/plotjuggler_plugins/ParserProtobuf/protobuf_factory.h:6:
In file included from /Users/renyunfan/Documents/plotjuggler_ws/src/PlotJuggler/plotjuggler_plugins/ParserProtobuf/protobuf_parser.h:12:
/Users/renyunfan/Documents/plotjuggler_ws/src/PlotJuggler/plotjuggler_plugins/ParserProtobuf/error_collectors.h:13:45: error: only virtual member functions can be marked 'override'
                const std::string& message) override;
                                            ^~~~~~~~
/Users/renyunfan/Documents/plotjuggler_ws/src/PlotJuggler/plotjuggler_plugins/ParserProtobuf/error_collectors.h:16:47: error: only virtual member functions can be marked 'override'
                  const std::string& message) override;
                                              ^~~~~~~~
/Users/renyunfan/Documents/plotjuggler_ws/src/PlotJuggler/plotjuggler_plugins/ParserProtobuf/error_collectors.h:31:45: error: only virtual member functions can be marked 'override'
                const std::string& message) override;
                                            ^~~~~~~~
/Users/renyunfan/Documents/plotjuggler_ws/src/PlotJuggler/plotjuggler_plugins/ParserProtobuf/error_collectors.h:34:47: error: only virtual member functions can be marked 'override'
                  const std::string& message) override;
                                              ^~~~~~~~
4 errors generated.
make[2]: *** [plotjuggler_plugins/ParserProtobuf/CMakeFiles/ProtobufParser.dir/ProtobufParser_autogen/mocs_compilation.cpp.o] Error 1
make[1]: *** [plotjuggler_plugins/ParserProtobuf/CMakeFiles/ProtobufParser.dir/all] Error 2
make: *** [all] Error 2
```

tonygon, 2024-04-22:
Same here.
I could only get all the way to this point with the comments from https://github.com/facontidavide/PlotJuggler/issues/947, following the discussion here
https://github.com/facontidavide/PlotJuggler/pull/950/files 

When are those changes going to be merged?

```diff
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -42,6 +42,18 @@ if(VCPKG_TOOLCHAIN)
     message(STATUS "BUILDING_WITH_VCPKG")
 endif()

+# Include PkgConfig module
+find_package(PkgConfig REQUIRED)
+
+# Use pkg_check_modules to find ZeroMQ
+pkg_check_modules(ZEROMQ REQUIRED IMPORTED_TARGET libzmq)
+
+# Include ZeroMQ directories
+include_directories(${ZEROMQ_INCLUDE_DIRS})
+# -------
+
 find_package(ament_cmake QUIET)
 find_package(catkin QUIET)

```

Thanks!

syundo0730, 2024-05-04:
I managed to build it using an older version of protobuf.

```
brew install protobuf@21
brew link protobuf@21 --overwrite
```

TalGlantz, 2024-05-21:
Still having issues compiling.
I modified the files based on: https://github.com/facontidavide/PlotJuggler/pull/950/files updates.

Tried to compile and got same error. What am I missing?

`glantz@Tals-MacBook-Air plotjuggler_ws % cmake -S src/PlotJuggler -B build/PlotJuggler -DCMAKE_INSTALL_PREFIX=install
cmake --build build/PlotJuggler --config RelWithDebInfo --target install
-- The C compiler identification is AppleClang 15.0.0.15000309
-- The CXX compiler identification is AppleClang 15.0.0.15000309
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - broken
CMake Error at /opt/homebrew/Cellar/cmake/3.29.3/share/cmake/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "/Library/Developer/CommandLineTools/usr/bin/cc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: '/Users/glantz/plotjuggler_ws/build/PlotJuggler/CMakeFiles/CMakeScratch/TryCompile-rqLMnY'
    
    Run Build Command(s): /opt/homebrew/Cellar/cmake/3.29.3/bin/cmake -E env VERBOSE=1 /usr/bin/make -f Makefile cmTC_cdead/fast
    /Library/Developer/CommandLineTools/usr/bin/make  -f CMakeFiles/cmTC_cdead.dir/build.make CMakeFiles/cmTC_cdead.dir/build
    Building C object CMakeFiles/cmTC_cdead.dir/testCCompiler.c.o
    /Library/Developer/CommandLineTools/usr/bin/cc   -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX14.4.sdk -MD -MT CMakeFiles/cmTC_cdead.dir/testCCompiler.c.o -MF CMakeFiles/cmTC_cdead.dir/testCCompiler.c.o.d -o CMakeFiles/cmTC_cdead.dir/testCCompiler.c.o -c /Users/glantz/plotjuggler_ws/build/PlotJuggler/CMakeFiles/CMakeScratch/TryCompile-rqLMnY/testCCompiler.c
    Linking C executable cmTC_cdead
    /opt/homebrew/Cellar/cmake/3.29.3/bin/cmake -E cmake_link_script CMakeFiles/cmTC_cdead.dir/link.txt --verbose=1
    /Library/Developer/CommandLineTools/usr/bin/cc  -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX14.4.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names /opt/homebrew/opt/qt@5/lib  CMakeFiles/cmTC_cdead.dir/testCCompiler.c.o -o cmTC_cdead
    ld: file cannot be mmap()ed, errno=22 path=/opt/homebrew/opt/qt@5/lib in '/opt/homebrew/opt/qt@5/lib'
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    make[1]: *** [cmTC_cdead] Error 1
    make: *** [cmTC_cdead/fast] Error 2
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:3 (PROJECT)

-- Configuring incomplete, errors occurred!
make: Makefile: No such file or directory
make: *** No rule to make target `Makefile'.  Stop.
`

lgkimjy, 2024-06-13:
@TalGlantz, I have the same issue. Could you let me know if you solved this compile error?

glebv, 2024-06-15:
I had the same issue, I used recipe from here https://github.com/facontidavide/PlotJuggler/issues/961#issuecomment-2068415634 but I had to add path to protobuf and mosquitto. It's not a fully correct solution but it solved my issue
```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 16477dd3..a240e057 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -39,6 +39,16 @@ if(VCPKG_TOOLCHAIN)
     message(STATUS "BUILDING_WITH_VCPKG")
 endif()

+
+# Include PkgConfig module
+find_package(PkgConfig REQUIRED)
+
+# Use pkg_check_modules to find ZeroMQ
+pkg_check_modules(ZEROMQ REQUIRED IMPORTED_TARGET libzmq)
+
+# Include ZeroMQ directories
+include_directories(${ZEROMQ_INCLUDE_DIRS})
+
 find_package(ament_cmake QUIET)
 find_package(catkin QUIET)

@@ -103,6 +113,8 @@ if(APPLE AND EXISTS /usr/local/opt/qt5)
     set(CMAKE_MACOSX_RPATH 1)
 elseif(APPLE AND EXISTS /opt/homebrew/opt/qt@5)
     list(APPEND CMAKE_PREFIX_PATH "/opt/homebrew/opt/qt@5")
+    list(APPEND CMAKE_PREFIX_PATH "/opt/homebrew/opt/protobuf@21")
+    list(APPEND CMAKE_PREFIX_PATH "/opt/homebrew/opt/mosquitto")
     set(CMAKE_MACOSX_RPATH 1)
 endif()
 ```

ksm-dg, 2024-11-24:
I'm on macOS 15.1 (Sequoia) and FWIW, I couldn't compile completely with either the latest protobuf or protobuf@21. If I uninstall all protobuf versions and make the changes [here](https://github.com/facontidavide/PlotJuggler/issues/961#issuecomment-2068415634) I can compile. That's fine for me because I don't need protobuf for now, but presumably this won't work for everyone.

Per the pending pull request that fixes building on Mac, some code changes are actually needed to remove the `override` errors reported: https://github.com/facontidavide/PlotJuggler/pull/1019/commits/7657fb56944b7dc14d037008192ac7f920feb90d#diff-f7cae0d361d127176e8a7c1887d71ef4043b24be0fb36ea7545a2545ae04c71b. The CI run for that same commit built successfully, but without protobuf, so I take it that protobuf isn't essential.

jeffmelville, 2025-01-03:
I think this would be resolved by https://github.com/facontidavide/PlotJuggler/pull/1012

cHemingway, 2025-01-11:
Using [the patch](https://github.com/facontidavide/PlotJuggler/issues/961#issuecomment-2068415634) to CMakeLists.txt listed above and installing an [earlier version of protobuf](https://github.com/facontidavide/PlotJuggler/issues/961#issuecomment-2094285187) fixed the build for me.