PlotJuggler/PlotJuggler #963: Custom series with sqrt not working

julianoes, 2024-04-11:
# Problem description

I'm trying to use the ground speed custom series but it doesn't accept the example for ground speed. When I use `math.sqrt()` instead of `sqrt()` it works.

[image]
[image]

# Answer these questions (or I will ignore your issue)

1. Linux mint with small changes on top of main: e4f678ba0c4c9f8bf56843f4fed8cade4a79efaa
2. Done
3. Don't know
4. N/A

Jaeyoung-Lim, 2024-07-22:
I can confirm that this is still a problem

julianoes, 2025-08-03:
I can't find that any pre-made snippets / custom series with `sqrt`, so I must have been confused and really `math.sqrt` is just the right way to do this. Closing.

closed 2025-08-03 by julianoes