PlotJuggler/PlotJuggler #977: Cannot open .BIN file because of toLower()

Georacer, 2024-05-23:
# Problem description

Would it be possible to remove the use of `toLower` here?
https://github.com/facontidavide/PlotJuggler/blob/main/plotjuggler_app/mainwindow.cpp#L1460

It prevents our plugin from readings logs ending with `.BIN`. Instead, they need to be renamed to end with `.bin`.

# Answer these questions (or I will ignore your issue)

1. Describe your platform / Operative System and the **version of PlotJuggler** you are using.
Ubuntu 22.04; PlotJuggler 3.9.1
2. Compile and try the latest version in the **main** branch and check if the problem persists.
Didn't do it, but I checked `main` and the line of code is still there.
3. Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".
This check is not applicable to our custom `.BIN` file.
4. If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.
Any Ardupilot dataflash log will do, but you'll need [the plugin](https://github.com/ArduPilot/plotjuggler-apbin-plugins) to open them.
If you still need to try it, let me know.

zdavkeos, 2024-06-30:
When we had a similar plugin problem, we just added the lower case extension name to the plugin's list of supported extensions. See the csv loader for an example:
 https://github.com/facontidavide/PlotJuggler/blob/2c51aa689a69c7f1a2d715fd62b9179850fc23ff/plotjuggler_plugins/DataLoadCSV/dataload_csv.cpp#L96

Georacer, 2024-07-01:
Hello @zdavkeos,

Unfortunately, this is not working for the reasons I explained above.
I tried it, for good measure, though.

As an example, a file called `test_file.CSV` cannot be opened by PlotJuggler, for the same reasons.

muramura, 2024-09-11:
THIS PR:
https://github.com/facontidavide/PlotJuggler/pull/1003

closed 2025-12-27 by facontidavide