PlotJuggler/PlotJuggler #982: Conan compile instructions for Windows are wrong

briha, 2024-06-12:
# Problem description

When trying to compile with conan on Windows, COMPILE.md suggests to use 
```cmd
conan install src/PlotJuggler --install-folder build/PlotJuggler ^
      --build=missing -pr:b=default
```

However, `--install-folder` is not a valid argument for conan.
```cmd
conan install: error: unrecognized arguments: --install-folder build/PlotJuggler
```

`--output-folder` is likely the correct command line argument.

# Answer these questions (or I will ignore your issue)

1. Describe your platform / Operative System and the **version of PlotJuggler** you are using.
Windows 11, Conan 2.4.1, the latest PlotJuggler commit.

3. Compile and try the latest version in the **main** branch and check if the problem persists.
I am unable to compile. That's the problem.

4. Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".
No applicable.
6. If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.
No applicable.

briha, 2024-08-08:
For anyone else having the same, or a similar problem, here's how compiling PlotJuggler on Windows worked for me. This is using Visual Studio 16 – 2019 (which is [version 192 in Conan](https://stackoverflow.com/a/78376946)).

```
conan install src/PlotJuggler --output-folder build/PlotJuggler --build=missing -pr:b=default -s compiler=msvc -s compiler.version=192

set CMAKE_TOOLCHAIN=%cd%/build/PlotJuggler/conan_toolchain.cmake

cmake -G "Visual Studio 16" -S src/PlotJuggler -B build/PlotJuggler -DBUILDING_WITH_CONAN=ON -DCMAKE_TOOLCHAIN_FILE=%CMAKE_TOOLCHAIN% -DCMAKE_INSTALL_PREFIX=%cd%/install -DCMAKE_POLICY_DEFAULT_CMP0091=NEW

cmake --build build/PlotJuggler --config Release --target install -j14
```

And the installer can then be created by installing the Qt Installer Framework using QT's maintenance tool and then:
```
xcopy src\PlotJuggler\installer installer\ /Y /S /f /z
xcopy .\build\PlotJuggler\bin\Release\*.* installer\io.plotjuggler.application\data /Y /S /f /z

C:\Qt\Qt5.12.12\5.12.12\msvc2017_64\bin\windeployqt.exe --release installer\io.plotjuggler.application\data\plotjuggler.exe
C:\Qt\Qt5.12.12\Tools\QtInstallerFramework\4.8\bin\binarycreator.exe  --offline-only -c installer\config.xml -p installer  PlotJuggler-Windows-installer.exe
```

Valle125, 2024-09-21:
Thank you very much. This worked for me.

closed 2026-01-06 by facontidavide