PlotJuggler/PlotJuggler #985: ros-rolling-plotjuggler-ros package not available for last month

bijoua29, 2024-06-20:
# Problem description

The ros-rolling-plotjuggler-ros package has not been released and available for more than a month. 

# Answer these questions (or I will ignore your issue)

1. Describe your platform / Operative System and the **version of PlotJuggler** you are using. 
rolling distro on Ubuntu 24.04
2. Compile and try the latest version in the **main** branch and check if the problem persists.
Yes, I am able to compile the latest version from source and it runs fine. But I would rather use a released binary instead of compiling from source.
3. Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".
N/A
4. If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.
N/A

christophfroehlich, 2024-08-24:
It was released this week
```
$ apt show ros-rolling-plotjuggler-ros
Package: ros-rolling-plotjuggler-ros
Version: 2.1.3-1noble.20240820.001424
```

bijoua29, 2024-08-25:
Thanks for the update. Closing this issue.

closed 2024-08-25 by bijoua29