PlotJuggler/PlotJuggler #988: ROS1 Ubuntu 24 compatibility

RintoDeVries, 2024-07-03:
Hi! 

HUGE thanks to all the contributors of plotjuggler, it's awesome. 

I was wondering about the ros1 compatibility of plotjuggler in ubuntu 24. Playing back data is easy, since I can convert to mcap and load those in plotjuggler, but the issue is about streaming live data with ros1 (noetic). Is there any possible way to do this? Or a workaround people here use? 

Thanks in advance!

lucasw, 2024-08-23:
I've built plotjuggler from source within a catkin workspace in 24.04 and it can load and plot and playback ros1 bags fine, there are some instructions in https://github.com/lucasw/ros_from_src/tree/ubuntu2404 (but there's a lot else besides building plotjuggler there, if you need help feel free to ask in a discussion or issue there).

What it can't do is play back ros1msg mcaps using `ROS Topic Re-Publisher`, it doesn't show any topics under `Select topics to be published`, that may just be a minor bug and deserves a dedicated issue.

facontidavide, 2026-01-06:
not supporting ROS1 anymore, therefore I will close this

closed 2026-01-06 by facontidavide