PlotJuggler/PlotJuggler #992: Crash on Start UDP Server

light92131, 2024-07-20:
# Problem description
Let me say, I'm sure this is a local problem of some sort; others seem to use UDP Server without trouble. But for me:

Launch PlotJuggler; select UDP Server in Streaming drop-down; click Start. PlotJuggler terminates with:
```
...
Aborted (Signal sent by tkill() 260017 1000)
Aborted (core dumped)
```
(full backtrace below)

# Answer these questions (or I will ignore your issue)

1. Describe your platform / Operative System and the **version of PlotJuggler** you are using:
  - Linux Mint 21, 64-bit x86; native install (no VM)
  - PlotJugger built from _main_ HEAD @ 47a61d76

3. Compile and try the latest version in the **main** branch and check if the problem persists.

See above; built with:
```
$ cd PlotJuggler/
$ mkdir build; cd build
$ cmake -DCMAKE_BUILD_TYPE=Debug  ..
$ make -j2
```
5. Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".

Behavior is the same with or without `-t`. But maybe I need to do something with the sample data before Starting?

6. If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.

No data source here; just `Start`ing the UDP Server.

Here is the backtrace output from PlotJuggler:
```
eric@LighthartEngineering:~/LEI/Clients/internal/Experiments-plotjuggler/PlotJuggler/build$ ./bin/plotjuggler
Qt: Session management error: Could not open network socket
Loading compatible plugins from directory:  "/home/eric/LEI/Clients/internal/Experiments-plotjuggler/PlotJuggler/build/bin"
"libDataLoadCSV.so is a DataLoader plugin"
"libDataLoadMCAP.so is a DataLoader plugin"
"libDataLoadULog.so is a DataLoader plugin"
"libDataStreamMQTT_Mosquitto.so is a DataStreamer plugin"
"libDataStreamSample.so is a DataStreamer plugin"  ...disabled, unless option -t is used
"libDataStreamUDP.so is a DataStreamer plugin"
"libDataStreamWebSocket.so is a DataStreamer plugin"
"libDataStreamZMQ.so is a DataStreamer plugin"
"libParserDataTamer.so is a MessageParser plugin"
"libParserLineInflux.so is a MessageParser plugin"
"libParserROS1.so is a MessageParser plugin"
"libParserROS2.so is a MessageParser plugin"
"libProtobufParser.so is a MessageParser plugin"
"libPublisherCSV.so is a StatePublisher plugin"
Type conversion already registered from type QString to type QwtText
"libToolboxFFT.so is a Toolbox plugin"
"libToolboxLuaEditor.so is a Toolbox plugin"
Type conversion already registered from type QString to type QwtText
"libToolboxQuaternion.so is a Toolbox plugin"
Number of plugins loaded:  16 

Loading compatible plugins from directory:  "/home/eric/.local/share/PlotJuggler"
Number of plugins loaded:  0 

qt.accessibility.atspi: Sending TextCaretMoved from object that does not implement text interface:  QAccessibleInterface(0x55d3cc26c230 name="use field as timestamp if available" role=Client obj=QLineEdit(0x55d3cc26a520)"focusable|invisible")
qt.accessibility.atspi: Sending TextCaretMoved from object that does not implement text interface:  QAccessibleInterface(0x55d3cc14f4c0 name="use field as timestamp if available" role=Client obj=QLineEdit(0x55d3cc26bdc0)"focusable|invisible")
qt.accessibility.atspi: Sending TextCaretMoved from object that does not implement text interface:  QAccessibleInterface(0x55d3cc61b630 name="use field as timestamp if available" role=Client obj=QLineEdit(0x55d3cc26d0e0)"focusable|invisible")
qt.accessibility.atspi: Sending TextCaretMoved from object that does not implement text interface:  QAccessibleInterface(0x55d3cc5e5dc0 name="use field as timestamp if available" role=Client obj=QLineEdit(0x55d3cc271840)"focusable|invisible")
^^^^^ Output prior to clicking Start

VVVV Output after clicking Start
terminate called after throwing an instance of 'std::out_of_range'
  what():  map::at
Stack trace (most recent call last):
#31   Object "/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5", at 0x7fc9a536ad6d, in 
#30   Object "/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7fc9a9918a2b, in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#29   Object "/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7fc9a9943306, in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*)
#28   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fc9a955ce39, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#27   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fc9aa04e712, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#26   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fc9aa0adfd4, in 
#25   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fc9aa0aad3f, in 
#24   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fc9aa054e46, in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool)
#23   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fc9a955ce39, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#22   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fc9aa056363, in QApplication::notify(QObject*, QEvent*)
#21   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fc9aa04e712, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#20   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fc9aa0914ed, in QWidget::event(QEvent*)
#19   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fc9aa1461e6, in QAbstractButton::mouseReleaseEvent(QMouseEvent*)
#18   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fc9aa145fc3, in 
#17   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fc9aa14439d, in 
#16   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fc9aa144115, in QAbstractButton::clicked(bool)
#15   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fc9a95944e4, in 
#14   Object "./bin/plotjuggler", at 0x55d3ca91801b, in 
#13   Object "./bin/plotjuggler", at 0x55d3ca9179d4, in 
#12   Object "./bin/plotjuggler", at 0x55d3ca99bde3, in 
#11   Object "./bin/plotjuggler", at 0x55d3ca98cfbb, in 
#10   Object "/home/eric/LEI/Clients/internal/Experiments-plotjuggler/PlotJuggler/build/bin/libDataStreamUDP.so", at 0x7fc99c06d40a, in UDP_Server::start(QStringList*)
#9    Object "/home/eric/LEI/Clients/internal/Experiments-plotjuggler/PlotJuggler/build/bin/libDataStreamUDP.so", at 0x7fc99c06cdb7, in 
#8    Object "/home/eric/LEI/Clients/internal/Experiments-plotjuggler/PlotJuggler/build/bin/libDataStreamUDP.so", at 0x7fc99c07325f, in std::map<QString, std::shared_ptr<PJ::ParserFactoryPlugin>, std::less<QString>, std::allocator<std::pair<QString const, std::shared_ptr<PJ::ParserFactoryPlugin> > > >::at(QString const&) const
#7    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fc9a911c49f, in std::__throw_out_of_range(char const*)
#6    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fc9a91254d7, in __cxa_throw
#5    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fc9a9125276, in std::terminate()
#4    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fc9a912520b, in 
#3    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fc9a9119b9d, in 
#2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fc9a8d6f7f2, in abort
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fc9a8d89475, in raise
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fc9a8ddd9fc, in pthread_kill
Aborted (Signal sent by tkill() 261951 1000)
Aborted (core dumped)
eric@LighthartEngineering:~/LEI/Clients/internal/Experiments-plotjuggler/PlotJuggler/build$
```
Here is a screenshot of Eclipse debug session that has additional info. Executing line 117 in `udp_server.cpp`:

[image]

Thanks for listening!

zdavkeos, 2024-07-26:
I'm guessing this is a settings issue. Try looking for the Qt settings file on your machine and changing the value for UDP::Protocol to "JSON".

It would still be nice to have it catch this, I will look into a fix.

light92131, 2024-07-29:
I did not have any luck finding a Qt settings file. But, I looked at lines 99-102 in `udp_server.cpp`. After clicking `Start` -- but before the crash -- we look for `UDP_Server::protocol` in `QSettings`. (I can't figure out where the `QSettings` file is :( )

Is that what you're referring to?

It looks like `QString protocol` defaults to "JSON" in any case. That's the value I see in `protocol` after line 100 executes.

Thanks for looking at this!

coremarine, 2024-07-29:
Same crash here, just my luck that I wanted to try out for the first time the UDP stream functionality today. My crash report looks to have some extra info, this is [version 3.9.2 installed from the AUR](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=plotjuggler) for reference.

```
[dsa@rockyraccoon ~]$ plotjuggler
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Loading compatible plugins from directory:  "/usr/local/bin"
"libDataLoadCSV.so is a DataLoader plugin"
"libDataLoadMCAP.so is a DataLoader plugin"
"libDataLoadParquet.so" :  "Cannot load library /usr/local/bin/libDataLoadParquet.so: (libparquet.so.1600: cannot open shared object file: No such file or directory)"
"libDataLoadULog.so is a DataLoader plugin"
"libDataStreamSample.so is a DataStreamer plugin"  ...disabled, unless option -t is used
"libDataStreamUDP.so is a DataStreamer plugin"
"libDataStreamWebSocket.so is a DataStreamer plugin"
"libDataStreamZMQ.so is a DataStreamer plugin"
"libParserDataTamer.so is a MessageParser plugin"
"libParserLineInflux.so is a MessageParser plugin"
"libParserROS1.so is a MessageParser plugin"
"libParserROS2.so is a MessageParser plugin"
"libProtobufParser.so" :  "Cannot load library /usr/local/bin/libProtobufParser.so: (libprotobuf.so.25: cannot open shared object file: No such file or directory)"
"libPublisherCSV.so is a StatePublisher plugin"
"libPublisherVideoViewer.so" :  "Cannot load library /usr/local/bin/libPublisherVideoViewer.so: (/usr/local/bin/libPublisherVideoViewer.so: undefined symbol: _ZN11VideoDialog16updateSliderUnitEv)"
Type conversion already registered from type QString to type QwtText
"libToolboxFFT.so is a Toolbox plugin"
"libToolboxLuaEditor.so is a Toolbox plugin"
Type conversion already registered from type QString to type QwtText
"libToolboxQuaternion.so is a Toolbox plugin"
Number of plugins loaded:  14

Loading compatible plugins from directory:  "/home/dsa/.local/share/PlotJuggler"
Number of plugins loaded:  0

terminate called after throwing an instance of 'std::out_of_range'
  what():  map::at
Stack trace (most recent call last):
#31 | Source "../glib/glib/gmain.c", line 4217, in g_main_context_dispatch_unlocked
      Source "../glib/glib/gmain.c", line 4152, in g_main_context_iterate_unlocked [0x7eb3d4b769e6]
#30   Source "../glib/glib/gmain.c", line 3344, in g_main_dispatch [0x7eb3d4b14ab8]
#29   Object "/usr/lib/libQt5WaylandClient.so.5.15.14", at 0x7eb3d1a52e34, in userEventSourceDispatch(_GSource*, int (*)(void*), void*)
#28   Source "kernel/qwindowsysteminterface.cpp", line 1169, in sendWindowSystemEvents [0x7eb3d671cdcc]
#27   Source "kernel/qguiapplication.cpp", line 2285, in processMouseEvent [0x7eb3d673ba8f]
#26   Source "kernel/qcoreapplication.cpp", line 1064, in notifyInternal2 [0x7eb3d62aba67]
#25   Source "kernel/qapplication.cpp", line 3640, in notify_helper [0x7eb3d6f56330]
#24   Source "kernel/qwidgetwindow.cpp", line 300, in event [0x7eb3d6faacc6]
#23   Source "kernel/qwidgetwindow.cpp", line 684, in handleMouseEvent [0x7eb3d6fa8c89]
#22   Source "kernel/qapplication.cpp", line 2622, in sendMouseEvent [0x7eb3d6f59df4]
#21   Source "kernel/qcoreapplication.cpp", line 1064, in notifyInternal2 [0x7eb3d62aba67]
#20   Source "kernel/qapplication.cpp", line 3084, in notify [0x7eb3d6f5c0d0]
#19   Source "kernel/qapplication.cpp", line 3640, in notify_helper [0x7eb3d6f56330]
#18   Source "kernel/qwidget.cpp", line 8671, in event [0x7eb3d6f8f7c5]
#17   Source "widgets/qabstractbutton.cpp", line 1045, in mouseReleaseEvent [0x7eb3d70482d1]
#16   Source "widgets/qabstractbutton.cpp", line 409, in click [0x7eb3d7048100]
#15   Source "widgets/qabstractbutton.cpp", line 416, in emitClicked [0x7eb3d7046b72]
#14   Source ".moc/moc_qabstractbutton.cpp", line 308, in clicked [0x7eb3d70441a4]
#13   Source "kernel/qobject.cpp", line 3951, in doActivate<false> [0x7eb3d62df836]
#12   Source "/usr/src/debug/plotjuggler/build/plotjuggler_app/plotjuggler_autogen/EWIEGA46WW/moc_mainwindow.cpp", line 524, in qt_metacall [0x6479f89be9d9]
        521:         return _id;
        522:     if (_c == QMetaObject::InvokeMetaMethod) {
        523:         if (_id < 68)
      > 524:             qt_static_metacall(this, _c, _id, _a);
        525:         _id -= 68;
        526:     } else if (_c == QMetaObject::RegisterMethodArgumentMetaType) {
        527:         if (_id < 68)
#11   Source "/usr/src/debug/plotjuggler/PlotJuggler-3.9.2/plotjuggler_app/mainwindow.cpp", line 3397, in on_buttonStreamingStart_clicked [0x6479f8a0ba02]
       3394:   ui->buttonStreamingStart->setEnabled(false);
       3395:   if (ui->buttonStreamingStart->text() == "Start")
       3396:   {
      >3397:     startStreamingPlugin(ui->comboStreaming->currentText());
       3398:   }
       3399:   else
       3400:   {
[... 15 lines cut ...]
        133:   if (res == QDialog::Rejected)
#8  | Source "/usr/src/debug/plotjuggler/PlotJuggler-3.9.2/plotjuggler_plugins/DataStreamUDP/udp_server.cpp", line 117, in at
    |   115:       }
    |   116:     }
    | > 117:     parser_creator = parserFactories()->at(selected_protocol);
    |   118:
    |   119:     if (auto widget = parser_creator->optionsWidget())
      Source "/usr/include/c++/14.1.1/bits/stl_map.h", line 562, in operator() [0x7eb39bfcf71a]
        559:       {
        560:    const_iterator __i = lower_bound(__k);
        561:    if (__i == end() || key_comp()(__k, (*__i).first))
[... cut ...]
#4    Source "/usr/src/debug/gcc/gcc/libstdc++-v3/libsupc++/eh_terminate.cc", line 48, in __builtin_abort [0x7eb3d5cadff9]
#3    Source "/usr/src/debug/gcc/gcc/libstdc++-v3/libsupc++/vterminate.cc", line 95, in __verbose_terminate_handler [0x7eb3d5c97b2b]
#2    Source "/usr/src/debug/glibc/glibc/stdlib/abort.c", line 79, in abort [0x7eb3d5a334c2]
#1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x7eb3d5a4c11f]
#0    Source "/usr/src/debug/glibc/glibc/nptl/pthread_kill.c", line 44, in __pthread_kill_implementation [0x7eb3d5aa53f4]
Aborted (Signal sent by tkill() 89667 1000)
Aborted (core dumped)
```

light92131, 2024-08-08:
@zdavkeos : I found the settings file, and added an entry to set UDP protocol to "JSON." No change.

I've attached a patch here that appears to fix the immediate issue. I can turn this into a PR if it looks appropriate.

[0001-Issue-992-Crash-on-Start-UDP-Server.patch.gz](https://github.com/user-attachments/files/16549045/0001-Issue-992-Crash-on-Start-UDP-Server.patch.gz)

closed 2024-11-10 by facontidavide

lu-maca, 2025-01-17:
Ciao @facontidavide, thanks for the big work. The same problem exists for ZMQ subscriber. I can't do proper debugging, but the line you changed in this PR is also in `datastream_zmq.cpp` (lines 74):
```cpp
// load previous values
QSettings settings;
QString address = settings.value("ZMQ_Subscriber::address", "localhost").toString();
QString protocol = settings.value("ZMQ_Subscriber::protocol", "JSON").toString();
QString topics = settings.value("ZMQ_Subscriber::topics", "").toString();
```

Is this easily fixable?

PR #1002 (2024-11-10): udp_server: fix crash when dialog opens for the first time
UDP server dialogue crashes when theres no protocol stored in settings, as the default is wrongly "JSON", but it should be "json".

This patch uses default "json" even when there is wrong protocol in settings to prevent further crashes.

Fixes #992