PlotJuggler/PlotJuggler #1006: MQTT subscriber 3.9.0 windows not work at all

matthijsfh, 2024-09-10:
# Problem description

Running Windows 10.

Just upgraded from 3.6.0 to 3.9.0

But now MQTT dialog does not list any of the (persistent) topics. Closed, run 3.6, showing all topics. (needed 3.9 to be able to select more than 1 topic from the list).

So back to 3.6.0 for now.

Valle125, 2024-09-17:
I can confirm the problem. I have the same issues on Win 11.

It seems that there are related issues on Linux. I can not connect to a local broker on Ubuntu 22. See screenshot below. It works flawlessly with MQTT Explorer. I'll try to debug the problem further and eventually submit a patch.

[image]

Valle125, 2024-09-19:
I've been debugging a bit today on my Linux  machine (Plotjuggler 3.9.1). When I open the MQTT plugin window and hit the connect/disconnect button repeatedly, Plotjuggler becomes stuck. The program flow seems to remain within mosquitto_connect_bind_v5() inside mqtt_client.cpp. I've tried to fix this by starting with a fresh mosquitto instance at each connection attempt, which seems to work. The program flow no longer locks up and I'm able to display data.

I'll test on windows in the next few days and submit a pull request when I'm confident with the patch.

briha, 2024-09-27:
My colleagues and I were having the same issue. I had solved it by just using the paho mqtt library instead, but #1009 seems like a more robust solution!

closed 2024-11-10 by facontidavide

matthijsfh, 2024-11-11:
Tested on windows 10 and it does seem to work. Thanks!

PR #1009 (2024-11-10): Fix MQTT plugin (#1006, #1008)
This pull request fixes issues #1006 and #1008.

On Linux the MQTT plugin freezes PlotJugger see issue #1008. The program flow seems to be stuck within mosquitto_connect_bind_v5() on the second connection attempt. This is solved by starting with a fresh mosquitto instance at each connection attempt.

On Windows PlotJuggler doesn't show any topics after establishing a connection, see issue #1006. mqtt_client.cpp does not check the return code of mosquitto_loop_start(), which may return MOSQ_ERR_NOT_SUPPORTED on Windows. This causes the plugin to assume that the co [... cut]
commit 78fb7a13 (2024-11-10): Fix MQTT plugin (#1006, #1008) (#1009)

* mqtt_client: fix bug where program flow gets stuck when calling mosquitto_connect_bind_v5.

The connection procedure starts with a fresh mosquitto instance each time. The instance will be destroyed on disconnect.
All mosquitto return codes are now checked during client configuration.

* mqtt_client: add debug messages and improve QMessageBox messages.

* mqtt_client: add mosquitto thread, fix #1006.

Threaded mode of libmosquitto may not be supported on windows (libmosquitto < 2.1). At some point libmosquitto has removed threaded support (mosquitto_loo [... cut]