PlotJuggler/PlotJuggler #1013: ROS2 topic subscriber is not connecting

manikandanhu2016, 2024-09-25:
# Problem description
while i am connecting plotjuggler with ROS2 topic i am getting he following error popup

**The plugin thrown the following exception: 
 failed to initialize rcl node: rcl node's rmw handle is invalid, at ./src/rcl/node.c:415**
here i have uploaded the screenshot..... 

[image]
 

# Answer these questions (or I will ignore your issue)
Operative System: Ubuntu 22.04 LTS
PlotJuggler version: 3.9.1
[image]

kindly help me to use this tool hope this could be very useful for plotting graph

manikandanhu2016, 2024-09-25:
kindly help me actually i need to plot graph from autoware. universe ROS topic for that i try to connect the plotjuggler but above error comes

manikandanhu2016, 2024-10-01:
i have solved by installing plugins for plotjuggler but new error coming kindly check the screenshot provide me a solution 
[image]

ivrolan, 2024-11-11:
Hi,
I had a similar problem when I installed PlotJuggler with snap. Installing it via apt like this:

```
sudo apt install ros-${ROS_DISTRO}-plotjuggler
```

And making sure to source your workspace before in the terminal that you run plotjuggler with

```
ros2 run plotjuggler plotjuggler
```

worked for me.

facontidavide, 2026-10-01:
Closing the original PJ3 3.9.1 installation report: the reporter confirmed the initial initialization error was resolved by installing the ROS plugins, and the maintainer identified the follow-up as a duplicate.

The working setup described above is the ROS distribution package (`sudo apt install ros-${ROS_DISTRO}-plotjuggler`), launched with `ros2 run plotjuggler plotjuggler` from a terminal where the ROS installation and workspace are sourced. Avoid mixing a standalone/Snap installation with incompatible ROS plugin binaries.

If an initialization failure is still reproducible with a current version, please open a focused report with the application/plugin versions, ROS distribution, installation method, and complete current error. PJ4 uses the separately packaged ROS2 Topic Subscriber extension with a matching per-distribution payload.

closed 2026-10-01 by facontidavide