PlotJuggler/PlotJuggler #1017: No actual data shown even though it is flowing in and displayed in the left column

deets, 2024-10-31:
# Problem description

I'm consuming JSON-formatted IMU data via ZMQ. The subscription process works just fine, and I've used PlotJuggler within the same project successfully before. However all of a sudden I do not get any actual graphs shown when dragging/dropping values into the view pane, even though I see the values update in the Timeseries List. The graphs are rendering with the time axis moving, just no data shown.

Every now and then I see some flickering in the graph views, but it doesn't look like the data, more like a rendering glitch. 

# Answer these questions (or I will ignore your issue)

1. I'm using Linux (ubuntu 22.40, Wayland, KDE), freshly built PJ version 3.9.2
2. Compile and try the latest version in the **main** branch and check if the problem persists -> **Did that, problem persists. f4f9d4192aee87fefac205fd48285b407f97e0b9**
3. Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples". -> **Did that, -t. Problem persists. I can see the data changing, but dragging it into the view pane doesn't make it show up, only scroll the timeline**

deets, 2024-10-31:
Ok, I rummaged through the dependencies, and found OpenGL. Disabled it. It works. Any hint as to what could be the root cause here?

facontidavide, 2026-09-20:
PlotJuggler 4 makes the OpenGL canvas a discoverable Preferences toggle (with a note that it applies to newly-created plots) plus a --disable-opengl launch flag, so you can switch to the software raster canvas without hunting through dependencies. If your driver still fails to draw curves on the GL canvas, that hardware-specific issue is a separate report.

---
*PlotJuggler 4 is now the official release and the default branch of this repo; PlotJuggler 3 is end-of-life and no longer receives features or fixes. If this still affects you in PJ4, please reopen or file a fresh issue — it will be looked at.*

closed 2026-09-20 by facontidavide