PlotJuggler/PlotJuggler #1030: Failed compiling on MacOS M2

c4stelo, 2024-12-18:
# Problem description
I followed the installation procedure provided with cmake commands and I can't compile plotjuggler 
cmake -S src/PlotJuggler -B build/PlotJuggler -DCMAKE_INSTALL_PREFIX=install                                                                   ─╯
cmake --build build/PlotJuggler --config RelWithDebInfo --target install

Describe the problem and the expected behavior. Use screenshots or a video, if applicable.

-- The C compiler identification is AppleClang 16.0.0.16000026
-- The CXX compiler identification is AppleClang 16.0.0.16000026
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - broken
CMake Error at /opt/homebrew/share/cmake/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: '/Users/c4stelo/FST/plotjuggler/build/PlotJuggler/CMakeFiles/CMakeScratch/TryCompile-l1xJUB'

    Run Build Command(s): /opt/homebrew/bin/cmake -E env VERBOSE=1 /opt/homebrew/bin/gmake -f Makefile cmTC_76cad/fast
    /opt/homebrew/bin/gmake  -f CMakeFiles/cmTC_76cad.dir/build.make CMakeFiles/cmTC_76cad.dir/build
    gmake[1]: Entering directory '/Users/c4stelo/FST/plotjuggler/build/PlotJuggler/CMakeFiles/CMakeScratch/TryCompile-l1xJUB'
    Building C object CMakeFiles/cmTC_76cad.dir/testCCompiler.c.o
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc   -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.2.sdk -MD -MT CMakeFiles/cmTC_76cad.dir/testCCompiler.c.o -MF CMakeFiles/cmTC_76cad.dir/testCCompiler.c.o.d -o CMakeFiles/cmTC_76cad.dir/testCCompiler.c.o -c /Users/c4stelo/FST/plotjuggler/build/PlotJuggler/CMakeFiles/CMakeScratch/TryCompile-l1xJUB/testCCompiler.c
    Linking C executable cmTC_76cad
    /opt/homebrew/bin/cmake -E cmake_link_script CMakeFiles/cmTC_76cad.dir/link.txt --verbose=1
    ld: file cannot be mmap()ed, errno=22 path=/opt/homebrew/opt/qt@5/lib in '/opt/homebrew/opt/qt@5/lib'
    cc: error: linker command failed with exit code 1 (use -v to see invocation)
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc  -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.2.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names /opt/homebrew/opt/qt@5/lib  CMakeFiles/cmTC_76cad.dir/testCCompiler.c.o -o cmTC_76cad
    gmake[1]: *** [CMakeFiles/cmTC_76cad.dir/build.make:103: cmTC_76cad] Error 1
    gmake[1]: Leaving directory '/Users/c4stelo/FST/plotjuggler/build/PlotJuggler/CMakeFiles/CMakeScratch/TryCompile-l1xJUB'
    gmake: *** [Makefile:133: cmTC_76cad/fast] Error 2

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:13 (PROJECT)

-- Configuring incomplete, errors occurred!
gmake: Makefile: No such file or directory
gmake: *** No rule to make target 'Makefile'.  Stop.

# Answer these questions (or I will ignore your issue)

1. Describe your platform / Operative System and the **version of PlotJuggler** you are using.
I am on MacOS running 15.2 Sequoia
2. Compile and try the latest version in the **main** branch and check if the problem persists.
Can't compile
3. Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".
Can't compile
4. If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.
Can't compile

rlgus94, 2025-01-23:
It appears that you’re facing similar issues. Here’s what I’ve encountered with version 3.10:
```
  Run Build Command(s): /opt/homebrew/bin/cmake -E env VERBOSE=1 /usr/bin/make -f Makefile cmTC_5c861/fast
  /Applications/Xcode.app/Contents/Developer/usr/bin/make  -f CMakeFiles/cmTC_5c861.dir/build.make CMakeFiles/cmTC_5c861.dir/build
  Building C object CMakeFiles/cmTC_5c861.dir/testCCompiler.c.o
  /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc   -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.2.sdk -MD -MT CMakeFiles/cmTC_5c861.dir/testCCompiler.c.o -MF CMakeFiles/cmTC_5c861.dir/testCCompiler.c.o.d -o CMakeFiles/cmTC_5c861.dir/testCCompiler.c.o -c /Users/kihyeonkim/plotjuggler_ws/build/PlotJuggler/CMakeFiles/CMakeScratch/TryCompile-zGBWdg/testCCompiler.c
  Linking C executable cmTC_5c861
  /opt/homebrew/bin/cmake -E cmake_link_script CMakeFiles/cmTC_5c861.dir/link.txt --verbose=1
  ld: file cannot be mmap()ed, errno=22 path=/opt/homebrew/opt/qt@5/lib in '/opt/homebrew/opt/qt@5/lib'
  cc: error: linker command failed with exit code 1 (use -v to see invocation)
  /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc  -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.2.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names /opt/homebrew/opt/qt@5/lib  CMakeFiles/cmTC_5c861.dir/testCCompiler.c.o -o cmTC_5c861
  make[1]: *** [cmTC_5c861] Error 1
  make: *** [cmTC_5c861/fast] Error 2
```

It just came out from `PROJECT(plotjuggler LANGUAGES C CXX VERSION 3.9.3)` in project `CMakeLists.txt` file.

The problem is the linker:
```
  ld: file cannot be mmap()ed, errno=22 path=/opt/homebrew/opt/qt@5/lib in '/opt/homebrew/opt/qt@5/lib'
```

It turns out that using
```sh
QT_HOME=$(brew --prefix qt@5)
export CPPFLAGS="-I $QT_HOME/include"
export PKG_CONFIG_PATH="$QT_HOME/lib/pkgconfig"
export LDFLAGS="-L $QT_HOME/lib"
```

Instead of
```sh
QT_HOME=$(brew --prefix qt@5)
export CPPFLAGS="-I $QT_HOME/include"
export PKG_CONFIG_PATH="$QT_HOME/lib/pkgconfig"
export LDFLAGS="$QT_HOME/lib"
```

worked for me.

I've asked for ChatGPT for this problem, and it says:

```text
When linking, the compiler expects flags like -L/path/to/lib to tell it “this directory is where you should look for libraries.” That’s how the linker knows to look inside /path/to/lib for .a and .dylib (or .so) files.

By contrast, a bare path like /path/to/lib (without -L) is interpreted as an object file (or library file) to be linked directly. Obviously, if a path is a directory rather than an actual file, the linker will fail with “file cannot be mmap()ed” or a similar error.

Hence:

-L/opt/homebrew/opt/qt@5/lib ⇒ “Look in /opt/homebrew/opt/qt@5/lib for libraries to link”
/opt/homebrew/opt/qt@5/lib ⇒ “Attempt to link the file named /opt/homebrew/opt/qt@5/lib” (which fails because it’s a directory).
```

Unfortunately, I’ve encountered another problem while building the source code (I’m currently searching for solutions…):
```
...
[ 94%] Building CXX object plotjuggler_plugins/ParserProtobuf/CMakeFiles/ProtobufParser.dir/ProtobufParser_autogen/mocs_compilation.cpp.o
In file included from /Users/kihyeonkim/plotjuggler_ws/build/PlotJuggler/plotjuggler_plugins/ParserProtobuf/ProtobufParser_autogen/mocs_compilation.cpp:2:
In file included from /Users/kihyeonkim/plotjuggler_ws/build/PlotJuggler/plotjuggler_plugins/ParserProtobuf/ProtobufParser_autogen/EWIEGA46WW/moc_protobuf_factory.cpp:10:
In file included from /Users/kihyeonkim/plotjuggler_ws/build/PlotJuggler/plotjuggler_plugins/ParserProtobuf/ProtobufParser_autogen/EWIEGA46WW/../../../../../../src/PlotJuggler/plotjuggler_plugins/ParserProtobuf/protobuf_factory.h:6:
In file included from /Users/kihyeonkim/plotjuggler_ws/build/PlotJuggler/plotjuggler_plugins/ParserProtobuf/ProtobufParser_autogen/EWIEGA46WW/../../../../../../src/PlotJuggler/plotjuggler_plugins/ParserProtobuf/protobuf_parser.h:12:
/Users/kihyeonkim/plotjuggler_ws/build/PlotJuggler/plotjuggler_plugins/ParserProtobuf/ProtobufParser_autogen/EWIEGA46WW/../../../../../../src/PlotJuggler/plotjuggler_plugins/ParserProtobuf/error_collectors.h:31:45: error: only virtual member functions can be marked 'override'
   31 |                 const std::string& message) override;
      |                                             ^~~~~~~~
/Users/kihyeonkim/plotjuggler_ws/build/PlotJuggler/plotjuggler_plugins/ParserProtobuf/ProtobufParser_autogen/EWIEGA46WW/../../../../../../src/PlotJuggler/plotjuggler_plugins/ParserProtobuf/error_collectors.h:34:47: error: only virtual member functions can be marked 'override'
   34 |                   const std::string& message) override;
      |                                               ^~~~~~~~
2 errors generated.
make[2]: *** [plotjuggler_plugins/ParserProtobuf/CMakeFiles/ProtobufParser.dir/ProtobufParser_autogen/mocs_compilation.cpp.o] Error 1
make[1]: *** [plotjuggler_plugins/ParserProtobuf/CMakeFiles/ProtobufParser.dir/all] Error 2
make: *** [all] Error 2
```

farhangnaderi, 2025-02-10:
> gmake[1]: *** [CMakeFiles/cmTC_76cad.dir/build.make:103: cmTC_76cad] Error 1
> gmake[1]: Leaving directory '/Users/c4stelo/FST/plotjuggler/build/PlotJuggler/CMakeFiles/CMakeScratch/TryCompile-l1xJUB'
> gmake: *** [Makefile:133: cmTC_76cad/fast] Error 2

Have you tried the latest main? MacOS builds should be ok now.

rlgus94, 2025-02-10:
> Have you tried the latest main? MacOS builds should be ok now.

Dear @farhangnaderi,
I have tested the latest build (10/02/25) on my macOS (following instructions on `COMPILE.md`), but I have encountered the following error:

```sh (base) kihyeonkim@MacBook-Pro plotjuggler_ws % cmake -S src/PlotJuggler -B build/PlotJuggler -DCMAKE_INSTALL_PREFIX=install

-- The C compiler identification is AppleClang 16.0.0.16000026
-- The CXX compiler identification is AppleClang 16.0.0.16000026
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - broken
CMake Error at /opt/homebrew/share/cmake/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: '/Users/kihyeonkim/plotjuggler_ws/build/PlotJuggler/CMakeFiles/CMakeScratch/TryCompile-1spTzY'
    
    Run Build Command(s): /opt/homebrew/bin/cmake -E env VERBOSE=1 /usr/bin/make -f Makefile cmTC_44e70/fast
    /Applications/Xcode.app/Contents/Developer/usr/bin/make  -f CMakeFiles/cmTC_44e70.dir/build.make CMakeFiles/cmTC_44e70.dir/build
    Building C object CMakeFiles/cmTC_44e70.dir/testCCompiler.c.o
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc   -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.2.sdk -MD -MT CMakeFiles/cmTC_44e70.dir/testCCompiler.c.o -MF CMakeFiles/cmTC_44e70.dir/testCCompiler.c.o.d -o CMakeFiles/cmTC_44e70.dir/testCCompiler.c.o -c /Users/kihyeonkim/plotjuggler_ws/build/PlotJuggler/CMakeFiles/CMakeScratch/TryCompile-1spTzY/testCCompiler.c
    Linking C executable cmTC_44e70
    /opt/homebrew/bin/cmake -E cmake_link_script CMakeFiles/cmTC_44e70.dir/link.txt --verbose=1
    ld: file cannot be mmap()ed, errno=22 path=/opt/homebrew/opt/qt@5/lib in '/opt/homebrew/opt/qt@5/lib'
    cc: error: linker command failed with exit code 1 (use -v to see invocation)
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc  -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.2.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names /opt/homebrew/opt/qt@5/lib  CMakeFiles/cmTC_44e70.dir/testCCompiler.c.o -o cmTC_44e70
    make[1]: *** [cmTC_44e70] Error 1
    make: *** [cmTC_44e70/fast] Error 2

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:3 (PROJECT)

-- Configuring incomplete, errors occurred! 
```

However, I successfully build at **PlotJuggler 3.10** with my recipe.
Here's how:

On Mac, the dependencies can be installed using [brew](https://brew.sh/) with the following command:

```shell
brew install cmake qt@5 protobuf@21 mosquitto zeromq zstd
```

It is mandatory to use `qt@5` and `protobuf@21`. Thus:

```shell
brew unlink qt
brew link qt@5
brew unlink protobuf
brew link protobuf@21
```

Add CMake into your env-vars to be detected by cmake

```shell
QT_HOME=$(brew --prefix qt@5) \
export CPPFLAGS="-I $QT_HOME/include" \
export PKG_CONFIG_PATH="$QT_HOME/lib/pkgconfig" \
export LDFLAGS="-L $QT_HOME/lib"
```

> Use `export LDFLAGS="-L $QT_HOME/lib"`: Linker - directory vs object file

Clone the repository into **~/plotjuggler_ws**:

```sh
git clone https://github.com/facontidavide/PlotJuggler.git ~/plotjuggler_ws/src/PlotJuggler
cd ~/plotjuggler_ws
```

Then compile using cmake:

```sh
# Option 1
cmake -S src/PlotJuggler -B build/PlotJuggler -DCMAKE_INSTALL_PREFIX=install
cmake --build build/PlotJuggler --config RelWithDebInfo --target install

# Option 2  (fallback)
cmake -S src/PlotJuggler -B build/PlotJuggler \
      -DCMAKE_INSTALL_PREFIX=install \
      -DCMAKE_PREFIX_PATH="/opt/homebrew/opt/protobuf@21;/opt/homebrew/opt/qt@5"
cmake --build build/PlotJuggler --target install
```

Revert original brew links:
```sh
# Revert brew link
brew unlink protobuf@21
brew link protobuf # Run one if you use protobuf@21 linking

brew unlink qt@5 # Run once you are done building to restore the original linking
brew link qt --override
```

Now, run PlotJuggler:
```sh
./install/bin/plotjuggler
```

closed 2025-02-10 by c4stelo