PlotJuggler/PlotJuggler #1044: plot categorical data

CarlSchwedes, 2025-02-07:
Hi there, I was wondered if plot juggler can also plot categorical data into its x/y line plots?

Imagine some strings to be encoded from real numbers
OBJ/0/attribute0 -> 0
OBJ/0/attribute1 -> 1
OBJ/0/attribute2 -> 2
...

Is there a way to directly print the attribute strings into a coordinate system (labels y-axis) instead of the number encodings?

facontidavide, 2025-03-03:
A plugin implementing this feature can be created, potentially

facontidavide, 2026-11-04:
Sorry for the delay in following up. Development will now focus exclusively on PlotJuggler 4 (PJ4).

PJ4 already offers categorical/state inspection through State Transitions: choose that view in an empty panel and drop a string, integer or boolean series onto it. We will not extend PJ3 numeric axes with categorical text labels. Closing the original PJ3 axis-extension request as not planned; State Transitions is the available PJ4 alternative.

closed 2026-11-04 by facontidavide