PlotJuggler/PlotJuggler #1060: Plugin exception "package not found" on ROS action feedback topics

BriceRenaudeau, 2025-04-11:
With the latest update the subscribtion on action feedback topic are not working anymore:

[image]

The package name seems to be cut 'docking::a' instead of 'docking::action::LocalServoing'.

[image]

facontidavide, 2025-04-29:
what do you need by "latest update"? Can you tell me which version worked and which one doesnt?

Also can you provide a sample rosbag to reproduce the error?

facontidavide, 2026-10-01:
PJ4 fix prepared in https://github.com/PlotJuggler/pj-official-plugins/pull/323.

Nested ROS action feedback uses `package::action`, but the schema builder removed five characters assuming every namespace ended in `::msg`. PJ4 therefore tried to resolve a truncated package and could not subscribe to action feedback.

Keep full `package/msg|action|srv/Type` names for typesupport lookup and traversal; emit canonical `package/Type` names in fields and `MSG:` sections so the existing ROS parser and Header/Quaternion handlers remain compatible. The same schema cache serves topic advertising and both subscription modes.

Fixes PlotJuggler/PlotJuggler#1060 for PJ4.

Validation: original lookup failure reproduced with a generated ROS Jazzy action; corrected schema generated through real typesupport and decoded by the published PJ4 ROS parser (`remaining=42.5`). All 21 ROS2 control-plane tests and 113 ROS parser integration tests pass. Proxy/plugin export checks, 47 script tests, SDK declaration/floor checks, pre-commit, and independent review pass. No original reporter bag was provided.

ROS2 streamer version: 1.1.1 → 1.1.2, with changelog entry. Build SDK remains 0.34.1; minimum SDK remains 0.28.0, with no new host contract.

closed 2026-11-03 by facontidavide