PlotJuggler/PlotJuggler #1068: Select which filename to export as CSV

ironcoder727, 2025-04-30:
I feel like this feature could be easy to implement and it would add a lot of value. I often want to compare two files with eachother, so I use the "Add prefix and merge" tick box. This lets me import several files and all is good. I then want to export these as separate CSV files - but when I select "export" the program doesn't let me select which file to export. This is a shame. After analyzing my data I instead need to open another plotjuggler instance and import one file at a time and export, remove that file from plotjuggler and then import, export the other.

Another idea. I have also noticed you can't select which timeseries you want to delete from your instance - the only choice you have is "Delete Points" or "Delete All". If there was a button for "Delete Selected" this would make this whole issue not as bad - but the nr 1 solution is obviously best. 

Tell me what you think of my idea and input.

Thx

ssys-zdavis, 2025-04-30:
I've had a similar request from my users. When exporting csv, they would like a "only selected timeseries" option. I'll take a look at the code and see what all is possible.

facontidavide, 2025-12-07:
on the roadmap