PlotJuggler/PlotJuggler #1070: CSV Datareader Plugin Timestamp Parsing Issue

anirudlappathi, 2025-05-02:
Error reading file | Couldn't parse timestamp
Parsing format: [None]
Time at line 1 : [2025-04-17T17:00:08]

I found that I am getting this error on the auto timestamp parsing.

#include <QString>
#include <QDateTime>
#include <QLocale>
#include <optional>
#include <iostream>
#include <iomanip>

std::optional<double> AutoParseTimestamp(const QString& str)
{
  bool is_number = false;
  QString str_trimmed = str.trimmed();
  double val = 0.0;

  // Support the case where the timestamp is in nanoseconds / microseconds
  int64_t ts = str.toLong(&is_number);
  const int64_t first_ts = 1400000000;  // July 14, 2017
  const int64_t last_ts = 2000000000;   // May 18, 2033
  if (is_number)
  {
    // check if it is an absolute time in nanoseconds.
    // convert to seconds if it is
    if (ts > first_ts * 1e9 && ts < last_ts * 1e9)
    {
      val = double(ts) * 1e-9;
    }
    else if (ts > first_ts * 1e6 && ts < last_ts * 1e6)
    {
      // check if it is an absolute time in microseconds.
      // convert to seconds if it is
      val = double(ts) * 1e-6;
    }
    else
    {
      val = double(ts);
    }
  }
  else
  {
    // Try a double value (seconds)
    val = str.toDouble(&is_number);
  }

  // handle numbers with comma instead of point as decimal separator
  if (!is_number)
  {
    static QLocale locale_with_comma(QLocale::German);
    val = locale_with_comma.toDouble(str, &is_number);
  }
  if (!is_number)
  {
    QDateTime ts = QDateTime::fromString(str, Qt::ISODateWithMs);
    if (ts.isValid())
    {
      return double(ts.toMSecsSinceEpoch()) / 1000.0;
    }
    else
    {
      return std::nullopt;
    }
  }
  return is_number ? std::optional<double>(val) : std::nullopt;
};

int main(int argc, char *argv[]) 
{
    QStringList testStrings = {
      "2025-04-17T17:00:08"
    };

    std::cout << std::fixed << std::setprecision(9);

    for (const QString& testStr : testStrings) {
        std::optional<double> result = AutoParseTimestamp(testStr);
        std::cout << "Input: [" << testStr.toStdString() << "] -> Parsed (seconds): ";
        if (result) {
            std::cout << *result << std::endl;
        } else {
            std::cout << "[Failed]" << std::endl;
        }
    }

    return 0;
}

I created this file to test the function that PlotJuggler uses itself and it works properly.

g++ tzparse.cpp -o timestamp_parser -lQt6Core -I/usr/include/x86_64-linux-gnu/qt6 -I/usr/include/x86_64-linux-gnu/qt6/QtCore -fPIC && ./timestamp_parser 
Input: [2025-04-19T22:07:29.568] -> Parsed (seconds): 1745125649.568000078

[image]

Valle125, 2025-07-10:
It works for me with custom format "yyyy-MM-ddThh:mm:ss.zzz" (Plot Juggler 3.10.9).

facontidavide, 2025-12-07:
I made a heavy refactoring (with some AI support) here: af2c5a16a6ca1826fe3625133c36f29db27be53f

Change should be available in 3.14.3

Please report if you have further issues